dresden-elektronik/deconz-rest-plugin #7239: Wrong version date shown in Phoscon

felker, 2023-09-16:
## Describe the bug

I access Phoscon from local IP.

The Phoscon > Settings > Gateway menu shows that I am running the latest release, 2.22.02. However, the release date written to the right of the version number is 2022-09-19, which seems to correspond to the 2.18.02 release. See below screenshot.

The deCONZ GUI "About" window correctly shows a 2023 copyright and the correct version (does not list the release date).

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

[image]

## Environment

 - Host system: (Raspberry Pi / PC / NAS) Raspberry Pi 3B running bullseye
 - Running method: Raspbian
 - Firmware version: 26720700
 - deCONZ version:2.22.02
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

felker, 2023-09-16:
I just noticed that coincidentally 2.23.01 was released today. I manually updated the package, and now no date appears after the version number--- so maybe this was a known problem in earlier versions?

[image]

Mimiix, 2023-09-16:
Hi

This is not the correct place to report this . Additionally the date is referenced to the firmware.

If you need any more information or still believe this is a bug: please use the forums.

Closing.

closed 2023-09-16 by Mimiix

felker, 2023-09-16:
Apologies, even though the screenshot was showing a Phoscon screen, I thought such a bug report fell under the purview of deCONZ since it is tied to the deCONZ release metadata. Phoscon bug reports are via the forum too?

The firmware that I am running seems to have been released on 2021-08-19, not the above date:
https://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26720700.bin.GCF