dresden-elektronik/deconz-rest-plugin #7255: Respect changes in deconz.service while installing update via apt

ErikOlufson, 2023-09-22:
## Describe the bug
I use multiple usb devices and therefore I have to specify the device for deconz. I modify  systemctls deconz.service and add  --dev. Whenever I update the package via apt this change gets lost.

```
# /lib/systemd/system/deconz.service
[Unit]
Description=deCONZ: ZigBee gateway -- REST API
Wants=deconz-init.service deconz-update.service
StartLimitIntervalSec=0

[Service]
User=1000
ExecStart=/usr/bin/deCONZ -platform minimal --dev=/dev/serial/by-path/pci-0000:00:14.0-usb-0:2:1.0 --http-port=80
Restart=on-failure
RestartSec=30
AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_KILL CAP_SYS_BOOT CAP_SYS_TIME

[Install]
WantedBy=multi-user.target
```
## Steps to reproduce the behavior
Install the deconz package via apt. Make changes in  deconz.service. Wait for an update and install new version of the packege. The previous changes will be lost.

## Expected behavior
The package setup should ask if the local file should be used or the packages version should be used and restart the service. The default should be the local file.

## Environment

 - Host system: Intel NUC
 - Running method: Debian 11
 - deCONZ version: 2.22.01-debian-buster-beta
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system?: Yes  rs232 USB serial adapter

senilio, 2023-09-22:
This is not how you are supposed to edit the systemd service. To avoid systemd service file getting overwritten on upgrades you need to use a systemd override file.

Basically you just go `sudo systemctl edit deconz`, edit the content and save. This will be saved to an `override.conf` which will take precedence over the config that comes with deCONZ

ebaauw, 2023-09-22:
For more background info, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/295 and https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2482.

SwoopX, 2023-09-22:
Sorry, but this is a question specific to an OS component and no deconz related bug.

Pointers have been given, therefore closing.

closed 2023-09-22 by SwoopX