dresden-elektronik/deconz-rest-plugin #7259: Tuya Water leak sensor

YvesSun23, 2023-09-24:
## Device

- Product name: Tuya Zigbee Water Sensor
- Manufacturer: _TZ3000_k4ej3ww2
- Model identifier: ATS0207
- Device type : Sensor
  
[image]

## Screenshots

### Basic
[image]

### Identify
[image]

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above
[image]
[image]
[image]
[image]
[image]

Mimiix, 2023-09-25:
> Sorry, I just saw that the device support has already been implemented. _TZ3000_TS0207_water_leak_sensor.json For some reason it is not displayed in Phoscon for me.

That is not the same one. The file you mention is not including your device as you can see https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0207_water_leak_sensor.json

@Smanar  can you check in :)?

YvesSun23, 2023-09-25:
Yes, I just opened the file in the DDF editor and the manufacturer code is not included.
Would it be enough to add the manufacturer code or do I have to create a completely new file?
I am currently trying to get to grips with the description of the DDF files from Dresden Elektronik.

YvesSun23, 2023-09-25:
Perfect, I'll test it at the weekend, unfortunately I'm away on business at the moment.
I will then let you know if it worked or not.

I have a question for understanding, as I have a few other cheap Tuya sensors, I would try to create my own DDF file.
If I understood correctly, I have to enter the values from the individual clusters in the DDF editor etc. and create a new file.
If I have done everything correctly and reconnect the sensor, then the DDF file is automatically loaded without me having to assign it manually?
If everything is entered correctly, then the sensor should also be displayed in the Phoscon App?

Translated with www.DeepL.com/Translator (free version)

Mimiix, 2023-09-25:
> Perfect, I'll test it at the weekend, unfortunately I'm away on business at the moment. I will then let you know if it worked or not.
>
> I have a question for understanding, as I have a few other [...]

Yes to most of it, but the last: Phoscon doesn't always show the sensors. However, it should show in the RestAPI.   Phoscon said they should show all devices now, but i'll have to see that 😅 

For help with a DDF, i recommend creating a device request and in there you'll get the help for that specific device. Additionally, you can also share the DDF for others to benefit :)

YvesSun23, 2023-09-25:
Good to know, if it is successfully included in deconz and not shown in Phoscon, then it certainly won't be visible in homkit either (Homebridge-deconz).
I'll try to enter the manufacturer in the existing file first and then I'll get back to you.
Thank you very much, that helped me a lot, slowly I understand deCONZ 🤪

Mimiix, 2023-09-25:
> Good to know, if it is successfully included in deconz and not shown in Phoscon, then it certainly won't be visible in homkit either (Homebridge-deconz). I'll try to enter the manufacturer in the [...]

It should show in Homekit after you added it succesfully in deCONZ. The DDF is a whitelist aswell, so if you haven't applied it,the device wont be in the RestAPI.

Smanar, 2023-09-27:
What is your OS ? You are not using docker ?

If it have worked, you just need to re-include the device, deconz will use the DDF to reconize it (easy way, it's possible too using the GUI but more manipulation)

Smanar, 2023-09-28:
So no problem on this OS.
You have edited the deconz file, so at next deconz update the file will be overwrited, so no danger.

YvesSun23, 2023-09-29:
Thanks again for the help and the DDF crash course.
The sensor was automatically integrated and assigned overnight without me having to do anything.

YvesSun23, 2023-09-29:
[image]

[image]

Smanar, 2023-09-29:
So it's ok I can make the PR to submit the device ?

YvesSun23, 2023-09-30:
Sorry, what is PR 🙈 
The device works fine, I have tested it and get a critical alarm in HomeKit if water is detected.

The only thing I am unsure of, is the battery capacity. I will swap the battery for a new one tomorrow and will see if the charge is displayed at 100%.
No idea how long the sensor has been in storage. But for just under 7€ (Ali...) the sensor works really well.

Smanar, 2023-09-30:
PR is that https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7271

The battery level is take from the cluster 0x0001 (power configuation), attribute 0x0021 (battery percent)

closed 2023-10-04 by manup

DickenSerm, 2024-01-25:
Hey, I also have a Tuya Water Sensor clone. Is it possible to record that too?

Manufacturer: _TZ3000_kstbkt6a

All other attributes and model identifier are exactly the same.

Smanar, 2024-01-25:
It's effectively a clone.
From Z2M 

`tuya.whitelabel('Aubess', 'IH-K665', 'Water leak sensor', ['_TZ3000_k4ej3ww2', '_TZ3000_kstbkt6a']),`

https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7549

DJF3, 2024-02-07:
Yesterday I received a [Onenuo water sensor](https://nl.aliexpress.com/item/1005005942082875.html) that is _just like the one_ at the top of this issue.
Except that mine has "Manufacturer Name" : `_TZ3000_mugyhz0q`

For now I have manually added it in the `_TZ3000_TS0207_water_leak_sensor.json` file.
After restarting Deconz this sensor works fine.

sciorty, 2024-02-09:
I have one of these sensors, but why don't I see the water state on the home assistant deconz device?
[image]
[image]
[image]

sciorty, 2024-02-09:
> I have one of these sensors, but why don't I see the water state on the home assistant deconz device?
> <img alt="Schermata 2024-02-09 alle 23 11 04" width="717" src="https://private-user-images.githubusercontent.com/6460914/303788020-09392f63-fb2d-43fd-b13d-b90bacf112b7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1MTcwODksIm5iZiI6MTcwNzUxNjc4OSwicGF0aCI6Ii82NDYwOTE0LzMwMzc4ODAyMC0wOTM5MmY2My1mYjJkLTQzZmQtYjEzZC1iOTBiYWNmMTEyYjcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDlUMjIxMzA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODk5MmE2N2RkNTA1MTI1NDk5MzYwMTgzMGJmMTQxNWVkN2Y4NzI3NDVmZjMyNTM4YWYwN2QyZDYzMmE3ZTU3YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.MUGwrLZQuzS8b4L3yhouEDELplkKwOXyllFoYxh2avk">
> <img alt="Schermata 2024-02-09 alle 23 11 14" width="721" src="https://private-user-images.githubusercontent.com/6460914/303788030-bab911ad-3a5f-48c3-832f-3e45200dad4b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1MTcwODksIm5iZiI6MTcwNzUxNjc4OSwicGF0aCI6Ii82NDYwOTE0LzMwMzc4ODAzMC1iYWI5MTFhZC0zYTVmLTQ4YzMtODMyZi0zZTQ1MjAwZGFkNGIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDlUMjIxMzA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzQ3MGYwMjAxMWVhMjBlOWYyODRjYmM1ZGM4OWYwMTNlODAwMGRmNjMyYmJiZWE2YzA2MjhmYWVkYjdiYWUyNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.4NlTJDNGHdg-xEJArGy8vNcP3uMOQ_Uy9-4URZS2plw">
> <img alt="Schermata 2024-02-09 alle 23 11 19" width="678" src="https://private-user-images.githubusercontent.com/6460914/303788034-9f637d7b-e5ad-496a-b8d1-beac4a32a553.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc1MTcwODksIm5iZiI6MTcwNzUxNjc4OSwicGF0aCI6Ii82NDYwOTE0LzMwMzc4ODAzNC05ZjYzN2Q3Yi1lNWFkLTQ5NmEtYjhkMS1iZWFjNGEzMmE1NTMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDlUMjIxMzA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWZjOTE3MTczMmZjZDU0NzhhOWUzZWJkOTI0MmY2NjM4NjAyNGMwZjBlMWM3OTUwNTU2ZDk0ZDU5YWIwM2JmYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.fzufxqhboo9JVKIgHSJl-s8O6WMVTF5_I28U_XwDPe0">

Well... I just had to reload the deconz integration a few times:
[image]

Smanar, 2024-02-10:
Can take time before the device send the first report.