dresden-elektronik/deconz-rest-plugin #7261: nous lz1

madmax5900, 2023-09-25:
- Product name: nous lz1
- Manufacturer: _TZ3000 _3vf5bnrm
- Model identifier: TS0011
- Device type : Please remove all unrelated device types. 
  - Light
[image]
[image]
[image]
[image]

On/Off

Mimiix, 2023-09-25:
Missing screenshots. Please add screens of all clusters.

madmax5900, 2023-09-25:
OK, is that okay?
[image]
[image]
[image]

[image]
t
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2023-09-25:
Looks good, thanks :)

Smanar, 2023-09-25:
Power source > battery !
It's a tuya joke ? It's a wired device ?

Can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZ3000_txpirhfq","_TZ3000_qmi1cfuq", "_TZ3000_ji4araar","_TZ3000_v7sopte0","_TYZB01_qeqvmvti", "_TZ3000_3vf5bnrm"],
  "modelid": ["TS0011", "TS0011", "TS0011","TS0011","TS0011", "TS0011"],
  "product": "1 gang switch module",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

madmax5900, 2023-09-25:
Hello!

Thanks for that.

The device is wired.

What do I do with it?

Smanar, 2023-09-25:
What is your OS ?
It's simple you just need to edit a text file, the devices\tuya\_TZ3000_TS0011_1gang_switch_module.json.
If you are a real OS can take less than 10mn, if you have docker, ... more ...

Mimiix, 2023-09-25:
The folder can be found here:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

After you need to rebind.

Smanar, 2023-09-25:
Docker, the official deconz docker image ? So will be easy > https://github.com/deconz-community/deconz-docker/issues/163#issuecomment-1366432917

>But unfortunately it doesn't work.

What happen ? The type haven't a big impact on result.

madmax5900, 2023-09-25:
I added the I to a group.
When I turn on the group it doesn't work.

If I click on the light itself it works.
[image]
[image]

Mimiix, 2023-09-25:
Hi,

We only do deconz and the rest api here , Not phoscon.

Smanar, 2023-09-25:
Can be from the DDF too, but for me I don't see problem on your device, you have the group cluster, and your device use normal cluster to work, the 0x0006.
You haven't error message ?

closed 2023-10-25 by github-actions[bot]