dresden-elektronik/deconz-rest-plugin #7263: Frient A/S Alarm motion sensor 2 MOSZB-154

linhgg3, 2023-09-27:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Alarm motion sensor 2

### Manufacturer

frient A/S

### Model identifier

MOSZB-154

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
[image]

SwoopX, 2023-09-27:
Oh, new toys from them. Should be fairly easy to intergrate. Will check their documentation later.

Smanar, 2023-09-27:
He realy look than the previous one the MOSZB-141 https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6523

I think a big part of the DDF can be re-used.

closed 2023-09-28 by manup