dresden-elektronik/deconz-rest-plugin #7266: Breathing Presence Detector - ZY-M100-1

jullit31, 2023-09-28:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

ZY-M100-1

### Manufacturer

_TZE204_qasjif9e

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Scenes
[image]

Mimiix, 2023-10-02:
Did you properly search the git?
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7184

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7183

Looks like its supported since 2.24.0.

Closing this issue.

closed 2023-10-02 by Mimiix

jullit31, 2023-10-02:
Ahh, thanks! I only searched for "ZY-M100-1", as that is the only identifier written on the box of my device...
So I'll just wait for 2.24.0.

Mimiix, 2023-10-02:
@jullit31  No problem!

In the future, you should search for the model id/ manufacturer name combo. 

also, you might be able to add the device already by getting its DDF. For help on that, i recommend reading https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet 

If you need any help, check the forums or our discord (as Github is not the correct place)