dresden-elektronik/deconz-rest-plugin #7276: Add Support for lumi.magnet.agl02 (Not lumi.senor_magnet.agl02)

iDragonfighter, 2023-10-04:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Aqara Door and Window Sensor T1

### Manufacturer

Aqara

### Model identifier

lumi.magnet.agl02

### Device type to add

None

### Node info

Hi,
there is already support for "lumi.senor_magnet.agl02". #4428 
But somehow my device shows. "lumi.magnet.agl02" (without the "sensor_").
Therefore it will not show in the app after connecting.

Like the other device it says "Aqara Door and Window Sensor T1" on the box.
The Box is beige other than my other devices wich came in an blue box.

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Mimiix, 2023-10-04:
Looks like a new one :)

Can you please add all screens?

SwoopX, 2023-10-04:
Sorry, but it's a duplicate of #4428

Mimiix, 2023-10-04:
> Sorry, but it's a duplicate of #4343

As far as I can see it is the sensor_magnet instead of magnet in your issue.

SwoopX, 2023-10-04:
> > Sorry, but it's a duplicate of #4343
> 
> As far as I can see it is the sensor_magnet instead of magnet in your issue.

You might want to look again closely 😉

Mimiix, 2023-10-04:
> > > Sorry, but it's a duplicate of #4343
> > 
> > 
> > As far as I can see it is the sensor_magnet instead of magnet in your issue.
> 
> You might want to look again closely 😉

You mean your edit on #4428? Now it's fine. Thanks

iDragonfighter, 2023-10-04:
Sorry, i usually don‘t use github and the terminology here and in the deconz Program is a bit too much for me 😂

You‘re right. 
It is the Same Device as mentioned in the request #4343
So if needed, you can take the Screenshots from there. 
But in the „commit?“ (in the changed Code) #4428
Ist says „sensor_magnet“ instead of „magnet“.

again, sorry if that was clear already.

closed 2023-11-06 by manup

iDragonfighter, 2024-01-10:
Thank you for implementing the T1 Sensor.
The sensor is now identified as "lumi.magnet.agl02". However, it seems to be malfunctioning as very few details are displayed in the Deconz App, and there is no Open/Closed state. 
[image]

SwoopX, 2024-01-10:
Have you re-paired it?

SwoopX, 2024-01-13:
@iDragonfighter Conbee 3?

iDragonfighter, 2024-01-13:
No, RaspBee II

SwoopX, 2024-01-13:
Hm, very odd. However, I also pleayed around with an E1 sensor and experienced the very same. The device was stuck in some strange state I have to say. It seemingly paired correctly (I had the sniffer running) but it just kept quiet while moving the magnet, literally zero response, but all that was with a Conbee 3.

I'll see that have another look with the T1 on Conbee 2 within the next few days.

sirinveen, 2024-01-24:
I've just bought this sensor and it works very well, just modify the DDF so that it takes on/off cluster. And also re-pair is always a good idea.

[image]

[image]

```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.magnet.agl02",
  "vendor": "Xiaomi",
  "product": "Aqara T1 open/close sensor MCCGQ12LM",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/xiaomi_mccgq12lm_t1_openclose_sensor.json",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500",
          "0xFCC0"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x01",
            "mf": "0x115f",
            "script": "xiaomi_battery.js"
          },
          "default": 0
        },
        {
          "name": "config/enrolled",
          "public": false,
          "default": 1
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/open",
          "refresh.interval": 5,
          "read": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 0,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          }
        }
      ]
    }
  ]
}
```

chc88, 2024-02-04:
The adjusted ddf by @sirinveen works for me

SwoopX, 2024-02-04:
The DDF has already been amended in this regard. However, I'd recommend to take code from here https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/xiaomi/xiaomi_dw-s03d_t1_openclose_sensor.json to prevent the device being hammered with unnecessary requests.

iDragonfighter, 2024-02-05:
I don't know what I'm doing wrong, but unfortunately, it still doesn't work for me.
Not with the DDF from @sirinveen, and not even with the DDF file provided by @SwoopX.
Same behavior as [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7276#issuecomment-1885802896).
I now believe that my sensor is simply broken...
I think I am going to toss it out.

But thanks for the support.