dresden-elektronik/deconz-rest-plugin #7280: Equation wire pilot heating

Mimic33, 2023-10-06:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Equation wire pilot heating

### Manufacturer

Adeo

### Model identifier

SIN-4-FP-21_EQU

### Device type to add

None

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering
[image]
[image]
[image]

### Electrical Measurement

### Green Power
[image]

Mimic33, 2023-10-06:
Hi, here an issue for a wire pilot heating module. It has been recognized in Deconz and Jeedom but I only have state/on and off in Jeedom. And the off option corresponds to the no frost protection on the heater, not the off option. Normally I would have 6 orders (Confort, Off, no frost protection, eco, ...). This module could also give the consumption. Thanks.

Mimic33, 2023-10-06:
Identify :
[image]

Groups :
[image]

Scenes :
[image]
[image]

Mimic33, 2023-10-06:
Do you want the OTAU also ?

Mimiix, 2023-10-06:
> Do you want the OTAU also ?

No , thats fine.

Please add the screens in the opening post, makes it easier to read :)

Mimic33, 2023-10-06:
OK, done :) Sorry it's my first time :)

Mimic33, 2023-11-02:
Hello, any news ? Thanks.

hcharbonnier, 2023-11-02:
News from Nodon (for zigbee2mqttsupport): "Integration is in progress. It will be done within a few days."

Mimic33, 2023-11-03:
OK thanks, and for Deconz, you know ?

Smanar, 2023-11-06:
lol, @pipiche38 you are using deconz too ?
On my side, it's not my priority as I already have problem with the Legrand one, the "fil pliote" is something "exotic" for not french user, and hard to integrate it without making a "special" stuff for it.

Mimic33, 2023-11-14:
OK thanks for the reply. Maybe ideas her https://github.com/Koenkk/zigbee-herdsman-converters/pull/6429

closed 2023-12-13 by github-actions[bot]

Mimic33, 2023-12-13:
This subject has been closed ?

Smanar, 2023-12-13:
Yep, by a bot after a period without reaction.
Can be re-opened, but I don't think someone have started something.

Mimiix, 2023-12-13:
@Smanar happy to be re open but what do we need ?

Smanar, 2023-12-14:
For me nothing, this device have a working mode too much exotic, it will be problematic in deconz and later in the HA plugin.
If someone want to spend some time, I think the DDF need to have something like

```
{
  "name": "config/mode",
          "values": [
              ["confort", 1], ["confort-1", 4], ["confort-2", 5], ["confort-2", 3], ["eco", 2], ["hors gel", 3], ["off", 0]
          ],
  "parse": {
    "fn": "zcl:cmd",
    "ep": 1,
    "cl": "0xFC00",
    "cmd": "0x00",
    "eval": "if (Attr.val == 0) { Item.val = 'off' } else if (Attr.val == 1) { Item.val = 'confort' } else { Item.val = 'confort-2' }"
  },
  "read": {
    "fn": "zcl:cmd",
    "ep": 1,
    "cl": "0xFC00",
    "cmd": "0x00",
    "eval": "if (Attr.val == 0) { Item.val = 'off' } else if (Attr.val == 1) { Item.val = 'confort' } else { Item.val = 'confort-2' }"
  },
  "write": {
    "fn": "zcl:cmd",
    "ep": 1,
    "cl": "0xFC00",
    "cmd": "0x00",
    "eval": "if (Item.val == 'off') { 0 } else if (Item.val == 'confort') { 1 } else { 3 }"
  }
}
```

Mimic33, 2023-12-30:
Thanks, and if I want to use it, what I have to do ?

Smanar, 2023-12-31:
Make a complete DDF.
For exemple to start

```
{
  "schema": "devcap1.schema.json",
  "ddfvalidate": false,
  "manufacturername": "Adeo",
  "modelid": "SIN-4-FP-21_EQU",
  "product": "Cable outlet",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0001",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201",
          "0xFC40"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/mode",
          "values": [["confort",1],["confort-1",4],["confort-2",5],["confort-2",3],["eco",2],["hors gel",3],["off",0]],
          "parse": {
            "fn": "zcl:cmd",
            "ep": 1,
            "cl": "0xFC00",
            "cmd": "0x00",
            "eval": "if (Attr.val == 0) { Item.val = 'off' } else if (Attr.val == 1) { Item.val = 'confort' } else { Item.val = 'confort-2' }"
          },
          "read": {
            "fn": "zcl:cmd",
            "ep": 1,
            "cl": "0xFC00",
            "cmd": "0x00",
            "eval": "if (Attr.val == 0) { Item.val = 'off' } else if (Attr.val == 1) { Item.val = 'confort' } else { Item.val = 'confort-2' }"
          },
          "write": {
            "fn": "zcl:cmd",
            "ep": 1,
            "cl": "0xFC00",
            "cmd": "0x00",
            "eval": "if (Item.val == 'off') { 0 } else if (Item.val == 'confort') { 1 } else { 3 }"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xFC00",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 1,
          "max": 3600,
          "change": "0x00000000"
        }
      ]
    }
  ]
}
```