dresden-elektronik/deconz-rest-plugin #7284: Ikea VindStyrka values of PM 2.5 no more provided

LeoeLeoeL, 2023-10-07:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

From 2.24.1 (I came from 2.23.1) Ikea Vindstyrka entities of PM 2.5 (numeric and "valutation") are no more available in Home Assistant.
In Deconz I see in 042A, MIN and MAX Values are "grey" now.

### Steps to reproduce the behavior

Update to 2.24.1
Reload Deconz Integration in HA
View the available entities of Vindstyrka

### Expected behavior

I expect to read again the 2 missing entities

### Environment

- Host system: Raspberry PI 3b
- Running method: Raspbian
- Firmware version: 26780700
- deCONZ version 2.24.1
- Device: ConBee II
- Do you use an USB extension cable: yes -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which? APC UPS

ebaauw, 2023-10-08:
We changed the `type` of the `sensors` resource reporting PM2.5 from ZHAAirQuality to ZHAParticulateMatter, see #7246.  Same on the STARKVIND air purifiers.

There was already another device carrying a ZHAParticulateMatter `sensors` resource, so I would expect API clients to be able to handle this.  However, it would seem that the deCONZ integration in HA needs to be updated, to reflect these changes.

Mimiix, 2023-10-08:
In that case, I guess we can close this issue :)

LeoeLeoeL, 2023-10-08:
Did you warn who has in charge the deCONZ integration about that?

Kane610, 2023-10-08:
First time I hear about it. Seems it might be a breaking change for users as well

Mimiix, 2023-10-08:
> First time I hear about it. Seems it might be a breaking change for users as well

It's been discussed in the internal channel.your part of.

Nevertheless, it's a beta. Wondering if Dresden announces breaking changes on stables or not.

As this is a different discussion, which doesn't belong here, I'll lock this up.

Feel free to have this discussion on the forums or discord.