dresden-elektronik/deconz-rest-plugin #7288: Tuya Millimeter Wave Presence Detector ZG-205Z

m-kuutti, 2023-10-09:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Human presence detector ZQ-205Z

### Manufacturer

_TZE200_2aaelwxk

### Model identifier

TS0225

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
[image]

[image]

BabaIsYou, 2023-10-11:
Seems a clone of this one #7223

Mimiix, 2023-11-27:
re-opening as theres someone else who can maybe test @BabaIsYou .

m-kuutti, 2023-11-27:
I can of course also test when there’s something to test, or do.

BabaIsYou, 2023-11-28:
What I meant i just that these two devices seems to be clone of each other. Then that a proposed DDF for one of them could probably fit for the second.

But I didn't proposed any DDF at this time ...

closed 2023-12-27 by github-actions[bot]

kwatt67, 2024-02-03:
Did you get a DDF that worked for this sensor as I tried the one referenced but that hasn't worked?

m-kuutti, 2024-02-03:
It hasn’t worked for me :(

jeanmi-git, 2024-02-18:
Hello,

I have received this presence sensor and it did not pair with my deConz conBee 2 zigbee dongle.
I running deconz 2.25.3 with docker on a Synology NAS.

I hope it will be compatible soon.

jeanmi-git, 2024-02-25:
> As there has not been any response in 28 days, this issue will be closed. **@ OP: If this issue is solved post what fixed it for you. If it is not solved, request to get this opened again.**

Could this issue be re-opened, please ?

megamacy, 2024-03-19:
> > As there has not been any response in 28 days, this issue will be closed. **@ OP: If this issue is solved post what fixed it for you. If it is not solved, request to get this opened again.**
> 
> Could this issue be re-opened, please ? I tested with the last version : 2.26, and sensor is still not discovered.

It is discovered, but not usable cause of missing compatible DDF

jeanmi-git, 2024-03-20:
I'm not familiar with the internal functionning of the deconz application.
How can we get a compatible DDF ?

megamacy, 2024-03-20:
Im new to this too. only got to it because the sensor is not working with deconz 

As far as i know, deconz need to be compatible (it is since v2.25 changelog) and then someone who knows how, need to create that ddf. A file that point the events to recognized commands. 
Once it is done, we can download it and put it to the "devices" directory of our deconz server and set DDF handling to "strict" for  enable load custom files, then pair it again

heres a ddf imported example, but its not working
[image]

Jupp2, 2024-03-20:
I am also still looking for a working ddf. I modified existing once but only came to a point that general object detection worked. But it was useless because too sensible. The Sensor generally has adjustable sensitivity and 2 types of object sizes or movement Levels it can differentiate. With Zigbee2MQTT all this works Like many other Sensors. I have the Feeling Zigbee2MQTT is better maintained.

megamacy, 2024-03-22:
> I am also still looking for a working ddf. I modified existing once but only came to a point that general object detection worked. But it was useless because too sensible. The Sensor generally has [...]

Can you share that ddf? i dont know how to get anything working on this sensor

Jupp2, 2024-03-22:
Will look for it when back home in about a week.

jeanmi-git, 2024-04-04:
Hello,
Did you find your DDF ?

Jupp2, 2024-04-04:
[ts0225.json](https://github.com/dresden-elektronik/deconz-rest-plugin/files/14865787/ts0225.json)
this is the DDF which works an a very basic level with my ZG-205Z. 
you may need to change the "manufacturername": "_TZE200_2aaelwxk" to the name you sensor has - they often differ slightly after the _ .

Jupp2, 2024-04-04:
Here is what it looks like in zigbee2mqtt: 
[image]
Any idea how to implement these settings and readings in Deconz by DDF (esp. "Motion state" and "Large motion detection distance" are relevant to use it as a human presence sensor)?

Tech4Ever, 2024-11-11:
Hi,

any progression on this issue? I received this sensor, too, but it´s not working.

gituser-rk, 2024-11-21:
Hello,
replace the file /usr/share/deCONZ/devices/tuya/ZY-M100_human_breathing_presence.json with the attached version. this works for me. I did a restart of the gateway after replacing the file. After that, I was able to integrate the sensor via Phoscon WEB-GUI.
There are several datapoints missing for configiring the sensor and the light level is also missing. Maybe it helps someone.
[ZY-M100_human_breathing_presence.json](https://github.com/user-attachments/files/17852718/ZY-M100_human_breathing_presence.json)