dresden-elektronik/deconz-rest-plugin #7289: AQARA Temperature and Humidity Sensor T1 - Wrong Icon and Pressure Measurement

mman, 2023-10-09:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I have just installed `deconz 2.24.1-beta` to verify https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7241.

For testing I have now two AQARA sensors sitting next to each other, one is the old non-T1 version, and the new T1.

The new T1 device is now properly recognised, and correctly reports temperature and humidity, however two issues remain:

1) The device is represented by a temperature icon, instead of multi sensor, see the attached screenshot with older AQARA non-T1 version.

[image]

2) The device incorrectly reports atmospheric pressure, in deconz GUI I can read 977, but web app reports 10 hPa.

[image]

### Steps to reproduce the behavior

Pair AQARA Multisensor T1.

### Expected behavior

The T1 multi sensor should be represented by multi sensor icon, as the previous non-T1 version and atmospheric pressure should be properly read out.

### Environment

- Host system: Raspberry Pi
- Running method: Raspbian
- deCONZ version: 2.24.1-beta
- Device: RaspBee II

SwoopX, 2023-10-09:
Closing as this is neither a bug nor relates to deconz and the information in the issue template has not been respected.

closed 2023-10-09 by SwoopX

SwoopX, 2023-10-10:
> [...] it then flies via deconz and displays in phoscon [...]

Indeed correct and what you're trying to highlight here are visualization issues in Phoscon. The template gives pretty clear guidance what to do in such cases, which is raising this in the forums.