dresden-elektronik/deconz-rest-plugin #7290: Namron Multisensor

le-clu, 2023-10-09:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Namron Zigbee Multisensor

### Manufacturer

Namron AS

### Model identifier

4512770

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

[image]

[image]

[image]

[image]

### Further relevant clusters

### Power Configuration
[image]

### IAS Zone
[image]

### Occupancy sensing
[image]

le-clu, 2023-10-09:
Hi. I already gave it a try to create a DDF file, and managed to see the sensors in Home Assistant. However, their values do not seem to update - and I don't really understand the motion sensor/IAS zone.

[namron-multisensor-4512770.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/12850096/namron-multisensor-4512770.json.txt)

Would really appreciate if anyone could help out - and of course make it an official device later on. :) 

Thanks!

BabaIsYou, 2023-10-10:
Well, nice try, but how to start ? ;-)

1. There are a lot of screenshots that are missing (clusters 0400, 0405, 0402, 0400, etc ...)
2. Why is "sleeper" to false ? Surprising if it's a battery powered device.
3.  Why did you detailed "attr/manufacturername" ? It's a standard Zigbee attribute
4.    Please change temp sensor to  {
```
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x03",
        "0x0402"
      ],
``` 
and let "state/temperature" without useless details

```
        {
          "name": "state/temperature"
        }
```
5. Change hum sensor with : 
```
     "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x04",
        "0x0405"
      ],
```
6. Change presence sensor with
```
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
```
and let "state/presence" without details

```
        {
          "name": "state/presence",
                  }
```

7. For light level we will see after the 0x0400 cluster screenshot to see if it returns lux or light level to adjust DDF. But for sure you can already change   ` "0x0106" `by `"0x0400"`
8. and last, but cetainly not least. there are no bindings into you're DDF, then may be some of the values won't be automatically reported. I suggest to read https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet

le-clu, 2023-10-16:
Hi. First of all thanks for your help! Appreciate your elaborate answer!

It was the first time for me to create a DDF but after your comments, I do understand it much better.

Implemented all changes as per your suggestions. 
Pasting additional screenshots below.

Regarding binding: I need to read a bit more about that - don't understand the concept yet, so I need some time.

One more generic question: how do I safely exchange DDFs between docker container and host? I run Home Assistant OS on RPI and made it work using 
`docker exec -it addon_core_deconz bash`
`docker cp [...]`
and it did the trick. BUT afterwards the container/add-on couldn't be stopped anymore. So after re-installing and restoring everything (I may have overreacted), I am now a bit defensive....

Thanks!

### Screenshots
**Light sensor 0400**
[image]

**Humidity sensor 0405**
[image]

**Temperature sensor 0402**
[image]

**IAS Zone 0500**
[image]

**Occupancy sensor 0406**
[image]

closed 2023-11-14 by github-actions[bot]

le-clu, 2023-11-14:
Hi @BabaIsYou 
Would you be able to have a look at my last reply please?

thanks!

BabaIsYou, 2023-11-14:
Hi,

sorry, very busy these last months. Can you display your last corrected DDF and saying what seems to work and what doesn’t ?

regarding docker and HA OS you’ll certainly find other posts on git, I’m not familiar with those.