dresden-elektronik/deconz-rest-plugin #7291: Haozee zigbee pir

basbeheertje, 2023-10-09:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Haozee zigbee pir

### Manufacturer

_TZ3000_mg4dy6z6

### Model identifier

TS0202

### Device type to add

Sensor

### Node info

It looks a lot like https://raw.githubusercontent.com/dresden-elektronik/deconz-rest-plugin/master/devices/tuya/ts0202_presence_sensor.json but is not present as manufacturer.

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Mimiix, 2023-10-10:
Missing a few screenshots. Can you please add them?

BabaIsYou, 2023-10-10:
> It looks a lot like https://raw.githubusercontent.com/dresden-elektronik/deconz-rest-plugin/master/devices/tuya/ts0202_presence_sensor.json but is not present as manufacturer.

You're right, may be then just try to add "_TZ3000_mg4dy6z6" into the "manufacturername" list, and aanother "TS0202" into the "modelid" list (you can try using DDF editor from the GUI) and I'm quite sure that it will do the trick.

basbeheertje, 2023-10-11:
> _TZ3000_mg4dy6z6

Yes it works

BabaIsYou, 2023-10-11:
Then you’re ready to make a pull request (PR) with the changes ;-)

closed 2023-11-09 by github-actions[bot]