dresden-elektronik/deconz-rest-plugin #7293: Philips Hue Xamento L does not accept effect colorloop anymore

bluemoehre, 2023-10-11:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Today my Xamento L was updated to firmware 1.93.7 and the colorloop effect does not work anymore.

API reports:
```
[
    {
        "error": {
            "address": "/lights/41/state/effect",
            "description": "Internal error, 951",
            "type": 901
        }
    }
]
```

I have to admit I didn't try it for a long time using this effect. It may already broke with a previous firmware.

In the HUE app I see the new effect "Prism" listed to this device. It wasn't there before.

If I try some of the new effects which already work for the GU10 bulbs, I get different errors:

```
[
    {
        "error": {
            "address": "/lights/41/state/effect",
            "description": "invalid value, candle, for parameter, effect",
            "type": 7
        }
    }
]
```

### Steps to reproduce the behavior

1. Pick a Hue which is running firmware 1.7x
2. Update firmware to 1.9x
3. Try to activate any effect

### Expected behavior

- colorloop is still working
- candle is now working
- fireplace is now working

### Environment

- Host system: PC
- Running method: Ubuntu / deCONZ docker
- Firmware version: 26720700
- deCONZ version: 2.22.02 / 19.9.2022
- Device: ConBee II

ebaauw, 2023-10-11:
> my Xamento L was updated to firmware 1.93.7

I'm not familiar with this particular model, and as far as I can tell, it's not listed in any DDF.  That would mean it's exposed by the legacy code, which would explain why the Hue manufacturer-specific effects, like `candle` and `prism` don't work for this model.  You can check the supported effects in the API response to a GET of the `lights` resource, in `capabilities/color/effects`.

Firmware 1.93.7 seems like an old version to me.  What's the _Image Type_ for this model?  Was the firmware upgraded by the Hue bridge or Hue app over BLE; or by deCONZ?

> the colorloop effect does not work anymore.

The internal error means that the REST API plugin has accepted the request, but didn't queue the corresponding Zigbee command.  My guess would be that the legacy code hasn't read the _Color Capabilities_ attribute, and consequenlty thinks the light doesn't support colorloop.  Could you check what the API reports for `colorcapabilties`, and if that's 0, could you try and read the _Color Contol_ cluster attributes from the GUI?

To add this model to the appropriate DDF, we need the screenshots as mentioned in the device request issue template.  Please make sure to read the attributes of each cluster.  If it supports any Hue-specific effects, it will carry a _Hue Effects_ cluster.

Mimiix, 2023-10-12:
I think we need a device request opened for this.

ebaauw, 2023-10-12:
> 0x011f

That's the same _Image Type_ as my the Hue Ensis and the outdoor light strip.  Latest firmware is version 1.104.2, but your version already carries the special effects.

> I think we need a device request opened for this.

Nah, we've got enough to add it to the existing DDF.

> Reading the attributes did not seem to change something.

Did you try to set the colorloop after that?

Mimiix, 2023-10-12:
> Nah, we've got enough to add it to the existing DDF.

Normally we'd require all screens. Or are the screens above enough?

ebaauw, 2023-10-12:
They are.  Hue lights are pretty standard.  They have a huge variance in _Model Indentifier_ values (sometimes even different values for the same device in a different colour), but only a few firmware builds.  And even different firmware builds act exactly the same, like this one, 0x011F and 0x0114, so they're exposed using the same DDF.

bluemoehre, 2023-10-12:
> That's the same Image Type as my the Hue Ensis and the outdoor light strip

Would make sense since all those devices use a lightstrip (internally).

>Nah, we've got enough to add it to the existing DDF.

I can imagine the Xamento M uses the same image too.

> Did you try to set the colorloop after that?

Actually I did but couldn't see any change.
**I tried now again and it worked** - seems to take some time to get announced everywhere.

API now reports the Xamento's _capabilities_ as:
```
{
    "alerts": [ "none", "select", "lselect"],
    "color": {
        "ct": {
            "max": 500,
            "min": 153
        },
        "effects": [ "none", "colorloop" ],
        "modes": [ "ct", "effect", "hs", "xy" ]
    }
}
```

Mimiix, 2023-10-14:
Please wait until the merge.

bluemoehre, 2023-11-05:
Dear Bot, plz keep this active as @Mimiix requested

closed 2023-11-06 by manup