dresden-elektronik/deconz-rest-plugin #7296: Device Request: SmartWings Roller Shades

Maarti, 2023-10-13:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

SmartWings Motorized Light Filtering Roller Shades

### Manufacturer

Smartwings

### Model identifier

WM25/L-Z

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

[image]

### Window Covering 

[image]

Maarti, 2023-11-05:
[Bump to prevent closing]

Mimiix, 2023-12-24:
@smanar Can you take a look?

Smanar, 2023-12-24:
Sure, can try this DDF pls ?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Smartwings",
  "modelid": "WM25/L-Z",
  "product": "SmartWings Roller Shades",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lift"
        },
        {
          "name": "state/open",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0102",
            "at": "0x0008",
            "eval": "Item.val = Attr.val < 100"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/alert"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 86400,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0102",
      "report": [
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 1,
          "max": 100,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3200,
          "max": 3600,
          "change": "0x00000002"
        }
      ]
    }
  ]
}
```

Maarti, 2023-12-24:
>From your capture it seem your percent battery is from 0 to 100 and not 200 like usual (or your battery was at 50% during your capture ?)

Yes I just recharged the battery fully to be sure and it shows `Battery Percentage Remaining: 100`.
[image]
____

It works, thank you!

The only problem is that in HomeAssistant the "open" and "close" functions are reversed. (However it correctly reports its current state).

I tried to re-pair it but same problem. This may be because I took the version of the blind where the motor turns in the other direction (the blind being away from the window). It's not a big deal, I would reverse it in Home Assistant.

Thanks again and happy holidays!

Smanar, 2023-12-25:
As you use the GUI, you have perhaps a command to reverse it.
Using the DDF I can reverse the return, so you will have return according to request (both reversed but more logic)

```
        {
          "name": "state/lift",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0102",
            "at": "0x0008",
            "eval": "Item.val = 100 - Attr.val"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "state/open",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0102",
            "at": "0x0008",
            "eval": "Item.val = Attr.val > 1"
          },
          "read": {
            "fn": "none"
          }
        },
```

Maarti, 2023-12-26:
In fact, its state is correctly returned ("closed" when physically closed and "open" when physically open).
But it's just the action of "opening" and "closing" that are reversed (the up arrow lowers the shade and the down arrow raises the shade).

When I try the fix above, their reported state becomes inverted.

But that's not a problem because I don't use the arrows but the "set_position()" service.

Thanks again!

Mimiix, 2023-12-26:
Hi, 

Please keep open until its merged into the code base for others to benefit.

Smanar, 2023-12-26:
>But that's not a problem because I don't use the arrows but the "set_position()" service.

Depend of POV, for me better to have return according to command, and reverse the whole system (return+command)