dresden-elektronik/deconz-rest-plugin #7298: GLEDOPTO GL-C-006 unresponsive but recover after pressing 0

ReX1983, 2023-10-13:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I have 4 dimmers GLEDOPTO GL-C-006 that become unresponsive within a few days of use.
They always become unresponsive while on (meaning that I turn them on and I can't turn them off anymore) and there are two **alternative** ways to resuscitate them:
1. power them off
2. select them in deConz and press zero (0).

If I power them off, after a while I can see their links in deConz. If I press 0, although they resuscitate, their links never come back as visible in deConz (even if they work for another couple of days).

The less I use these lights, the longer they remain online. If I don't use them, they work for weeks. If I use them often, they last 1-2 days.

These 4 devices are members of a deConz group. I control the group from Home Assistant.

They are connected to the coordinator via Ikea Tradfri repeaters/range extenders. There are 3 range extenders within 5 meters, but the situation has only slightly improved since I added such repeaters.

The GLEDOPTO are very close to each other (less than 4 meters away.
Notably, although these GLEDOPTO GL-C-006 are flagged as repeaters, they don't connect to anything, they act as end devices.

### Steps to reproduce the behavior

1. Use the devices as intended (switch them on/off and control brightness) for a few days

### Expected behavior

The devices remain responsive

### Screenshots

[image]

### Environment

- Host system: Raspberry PI 4
- Running method: Home Assistent deCONZ Add-on
- Firmware version: (0x26720700)
- deCONZ version: 2.22.2
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? no

bluemoehre, 2023-10-14:
I wonder if that problem also exists if the GLEDOPTOs are in range of the bridge. Your topology looks like it would make sense to use a second coordinator, which uses a (wired) network connection.

ReX1983, 2023-11-05:
Can anyone look into this? Thanks

Mimiix, 2023-11-27:
@Smanar You have a idea?

aihbrb, 2023-12-22:
I have the same issue with the GLEDOPTO GL-C-006 and a Conbee II (all updates) with Deconz (latest version). Sometimes it works for some days. Sometimes only a few hours.

ReX1983, 2023-12-22:
> I have the same issue with the GLEDOPTO GL-C-006 and a Conbee II (all updates) with Deconz (latest version). Sometimes it works for some days. Sometimes only a few hours.

I moved to zigbee2mqtt, due to this issue and a few others that were never looked at. The gledopto dimmers work flawlessly now.

Paule-Panter, 2023-12-27:
> GLEDOPTO GL-C-006

thx for this feedback

bluemoehre, 2023-12-29:
I now just bought GLEDOPTO GL-C-002P & GLEDOPTO GL-C-008P. Let's see if I experience the same issues and can support getting these solved.

Paule-Panter, 2023-12-29:
I use a

> Hersteller GLEDOPTO
> Modell GL-C-008P
> Version 10376801

and a

> Hersteller GLEDOPTO
> Modell GL-C-007P
> Version 10376801
 
in my zigbee network with a conbee II. They'll work without any issues the last years. In ny opinion they have also a very good antenna.

bluemoehre, 2023-12-29:
@Paule-Panter at first I can tell you my firmware is 20351203. On their website they seem to offer "something" for OTAU. I am not sure how to use that file, but maybe you find out.

Paule-Panter, 2023-12-29:
Thx for this notice, but I'll this running as it is. Never change a running system ;-)

closed 2024-01-28 by github-actions[bot]