dresden-elektronik/deconz-rest-plugin #7299: All devices in phoscon are not reachable

Artur-Fedjukevits, 2023-10-13:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

All devices are not reachable. In Phoscon all devices are greyed out. Can not control any of them.

### Steps to reproduce the behavior

Probably it happened when I tried to put stick to another usb port. After I put it back but nothing helping. Tried to restart PC, restart deconz, restart conbee2...

### Expected behavior

I don't want to reset all the lights. There is too many.

### Environment

- Host system: (PC)
- Running method: (Ubuntu)
- Firmware version: (26780700)
- deCONZ version (not Home assistant Addon version!): (2.14.01 / 06.02.2022)
- Device: (ConBee II)
- Do you use an USB extension cable: (yes) 
- Is there any other USB or serial devices connected to the host system? If so: Which? sonoff zigbee 3.0 usb dongle plus

### deCONZ Logs

21:52:36:438 CTRL skip creating node 0x8CF681FFFEBBEAE9 while permit join is disabled
21:52:37:581 Idle timer triggered
21:52:37:581 Force read attributes for ZHAPressure SensorNode Multi Sensor
21:52:40:398 New websocket 192.168.10.118:38876 (state: 3)
21:52:41:081 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
21:52:41:081 apsUseExtPanid is 0xEBD66273F80D7F5B but should be 0, start reconfiguration
21:52:41:081 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF069331, start reconfiguration
21:52:41:081 Skip automatic channel change, TODO warn user
21:52:42:581 Idle timer triggered
21:52:42:581 Force read attributes for ZHATemperature SensorNode Multi Sensor (childrens room)
21:53:31:081 tcAddress is 0x0000000000000000 but should be 0x00212EFFFF069331, start reconfiguration
21:53:31:081 Skip automatic channel change, TODO warn user

Artur-Fedjukevits, 2023-10-13:
more logs:

22:08:51:081 apsUseExtPanid is 0xEBD66273F80D7F5B but should be 0, start reconfiguration

Mimiix, 2023-10-13:
Clearly not a bug.

Ask for help on the forums.