dresden-elektronik/deconz-rest-plugin #7301: Zemismart 2 gang switch _TZ3000_lupfd8zu

kuszek, 2023-10-16:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Zemismart Tuya 2 gang no neutral switch

### Manufacturer

_TZ3000_lupfd8zu

### Model identifier

TS0012

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]
[image]

### Basic

[image]

### Further relevant clusters

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### On/Off
[image]
[image]

### Tuya specific
[image]

### OTAU
[image]

kuszek, 2023-10-16:
This device might be releated to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6870 or https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6798

Smanar, 2023-10-16:
Yeah ^^ probably. Can make a try just add the device to the list.
There is 2 kind of 2 gangs devices, the locked and the unlocked. Can try first the unlocked one

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_lupfd8zu",
  "modelid": "TS0012",
  "product": "Tuya 2 gangs wired without neutral",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        },
        {
          "name": "config/tuya_unlock"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

kuszek, 2023-10-16:
Thank you for such a quick response. It works :)

Smanar, 2023-10-16:
Nice ^^
Do you want to make the PR or want I make it ?

I think we can create a new file, because all other are for locked device. So a file called devices/tuya
/TS0012_2gang_wired_not_locked.json for exemple.

And need to replace TYPE_ON_OFF_OUTPUT by TYPE_ON_OFF_LIGHT (I have copy/paste a switch module for test)

kuszek, 2023-10-17:
I haven't created a PR yet so it would be best if you could do it.
Output worked fine for me. In Phoscon it shows up as a light and in Home Assistant it shows up a switch. It's the same for all my Zemismart switches, but if you want to fix I have no issue with that.

Smanar, 2023-10-17:
>Output worked fine for me. In Phoscon it shows up as a light and in Home Assistant it shows up a switch. It's the same for all my Zemismart switches, but if you want to fix it, I have no issue with that.

No, it don't cause issue, it's more "cosmetic"

PR done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7304

closed 2023-11-06 by manup