dresden-elektronik/deconz-rest-plugin #7303: Tuya Fingerbot Plus

Greymute, 2023-10-17:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Fingerbot Plus

### Manufacturer

_TZ3210_j4pdtz9v

### Model identifier

TS0001

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

[image]
[image]

Greymute, 2023-10-17:
I tried this (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7094) DDF but it didn't work...

Smanar, 2023-10-17:
Hello, you have a clone, have you tried the DDF on your link but with adding your model ?

```
  "manufacturername": ["_TZ3210_dse8ogfy", "_TZ3210_j4pdtz9v"],
  "modelid": ["TS0001", "TS0001" ],
```

Greymute, 2023-10-17:
`You have tried a re-inclusion ?`
Yes. It appears in Deconz and Phoscon, but I can't trigger it (aka turn it on). When I click on any of the "exec"-buttons in the On/off-Cluster, it works. Don't know if that means anything.
[image]

`What is your OS ?`
My OS is Windows 10 (running in a VM)

`If you select the node than "edit DDF" after a right clic, it's the good DDf that appear ?`
Yes, when I select "edit DDF" the correct DDF appears.

I'm running iobroker with the deconz-plugin. There I get several datapoints, but none of them seems to do anything:
Lights:
[image]
Sensors:
[image]

Smanar, 2023-10-18:
iobroker generate all command ? even not used on the device ? Because according to the DDF you need to have a config/delay on the sensor and nothing for dim in light.

>When I click on any of the "exec"-buttons in the On/off-Cluster, it works. Don't know if that means anything.

Yes, the DDF use the tuya cluster, not the on/off cluster, if it work using the GUI can try this DDF instead

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_dse8ogfy",
  "modelid": "TS0001",
  "vendor": "Tuya",
  "product": "Fingerbot",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
   {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 105, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name":"config/delay",
          "read":{
            "fn":"none"
          },
          "write":{
            "dpid":103,
            "dt":"0x2b",
            "eval":"Item.val;",
            "fn":"tuya"
          },
          "parse":{
            "dpid":103,
            "eval":"Item.val = Attr.val;",
            "fn":"tuya"
          },
          "default":1
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

I have let the tuya unlock sequence, and not added bind, so you can use issue on state return, but with this one, when you will use the state/on it will use now the cluster 0x0006.

Greymute, 2023-10-18:
Tried it with you DDF (changed the manufacturename to my model )and added it again in Phoscon. Now it works as it should :-D

[image]

The "switch" Datapoint acts as a toggleswitch, so every change (true -> false and vice versa) triggers the switch.

Smanar, 2023-10-19:
But your return is strange.
It mean some device work with tuya cluster and other with normal clusters ? It depend of what ? model/version/meteo/luck ?

Greymute, 2023-10-19:
I really don't know, this is all new for me :-\ 

On my end, Tuya-Devices always seem to be problematic in some ways. I have a Nous A1Z Smart Plug that is supportet by Phoscon but does some unexpected things - like not having a on/off-datapoint that ist directly accessible.
Most of the time you get several devices that are identical on the outside but have different firmware/modelnames/whatever that makes it kind of a lottery to get a device that's working right away with the DDF you might have :-(

Greymute, 2023-10-20:
OK, new problem:
I had to restart the PC that's running Deconz and the Fingerbot disappeared from Deconz :-(
When I try to re-add the device it only appears as "Battery" with the related datapoints. The Switch-Part doesn't appear....

closed 2023-11-18 by github-actions[bot]

mrrg1004, 2024-06-14:
Hey all,

I have the same issue that only the device only appears as "battery". I'm using deconz headless so I can't check the gui. Could you quickly help me in which folder I need to put the ddf? 

thank you in advance!

Smanar, 2024-06-14:
If I m right, without DDF the device is not reconised at all, so for me your DDf is fine.
And if the DDF is working, you will have both, can you check direclty in the API ? using Phoscon/help/API Information/light

mrrg1004, 2024-06-17:
Unfortunaly it's not visible in Phoscon/help/API Information/light

Smanar, 2024-06-18:
Ha, can you share the one in Phoscon/help/API Information/sensor ? it's the battery device.

mrrg1004, 2024-06-20:
Hey for sure,

"37": {
        "config": {
            "on": true,
            "reachable": true
        },
        "etag": "7d51ac3fd2977dce6065effd2a5472c6",
        "lastannounced": "2024-06-15T08:52:45Z",
        "lastseen": "2024-06-20T01:01Z",
        "manufacturername": "_TZ3210_j4pdtz9v",
        "modelid": "TS0001",
        "name": "Battery 37",
        "state": {
            "battery": null,
            "lastupdated": "none"
        },
        "swversion": "1.0.8",
        "type": "ZHABattery",
        "uniqueid": "a4:c1:38:94:90:0e:1c:51-01-0001"

I'm running deconz on 2.27.2 but had the same issue in 2.26.x.

Smanar, 2024-06-20:
Ha it seem you are not using the good DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3210_dse8ogfy_fingerbot.json
On this entry you need to have too "config/delay" but except that, all seem fine.

But long time this version was corrected (7 month) so you need to have the good one.

It's a new device, or you have already try to include it long time ago ?

Do you have access to the GUI ?