dresden-elektronik/deconz-rest-plugin #7307: Aqara Plug measurement & metering reports wrong values

bluemoehre, 2023-10-20:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I was tracking down an issue in Homeassistant where I can see the measurement / metering values of Aqara Plugs multiplied by 10x / 1000x. 
Now I checked Phoscon App and the same issue is there. 

**API reports wrong value in field `power`**.
The value should be divided by the factor which is listed in ** `Electrical Measurement / AC Power Divisor`.

```
{
  "config": {
    "on": true,
    "reachable": true,
    "temperature": 0
  },
  "ep": 1,
  "etag": "ca1e04aa15eaec5a3dc55c33adfd4d22",
  "lastannounced": "2023-10-16T22:13:25Z",
  "lastseen": "2023-10-20T00:09Z",
  "manufacturername": "LUMI",
  "modelid": "lumi.plug.maeu01",
  "name": "Aqara Plug 69",
  "state": {
    "current": 0,
    "lastupdated": "2023-10-20T00:09:16.057",
    "power": 440,
    "voltage": 0
  },
  "swversion": "12-06-2021",
  "type": "ZHAPower",
  "uniqueid": "54:ef:44:10:00:70:ff:fe-01-0b04"
}
```

**API reports wrong value in field `consumption`**.
The value should be divided by the factor which is listed in ** `Simple Metering / Divisor`.

```
{
  "config": {
    "on": true,
    "reachable": true,
    "temperature": 0
  },
  "ep": 1,
  "etag": "013182b778b257feb2ceb8bca0017208",
  "lastannounced": "2023-10-16T22:13:25Z",
  "lastseen": "2023-10-20T00:36Z",
  "manufacturername": "LUMI",
  "modelid": "lumi.plug.maeu01",
  "name": "Aqara Plug 68",
  "state": {
    "consumption": 308890,
    "lastupdated": "2023-10-20T00:36:49.174",
    "power": 0
  },
  "swversion": "12-06-2021",
  "type": "ZHAConsumption",
  "uniqueid": "54:ef:44:10:00:70:ff:fe-01-0702"
}
```

**Other values may need to be adjusted accordingly.**

### Steps to reproduce the behavior

1. Go to "Sensors" page
2. Choose an Aqara Plug (lumi.plug.maeu01)
3. See your smartphone charger pulling 440 Watts ;P

### Expected behavior

Correct value `40.0` is reported.
Other plugs / manufacturers may also choose different precisions.

### Screenshots

[image]

[image]

### Environment

- Host system: PC)
- Running method: Ubuntu / Docker
- Firmware version: 26720700
- deCONZ version: 2.22 & 2.24.1-beta
- Device: ConBee II
- Do you use an USB extension cable: yes

SwoopX, 2023-10-20:
Duplicate of #6251. Please use the search next time, it feels the tick was just set to get issue through.

closed 2023-10-20 by SwoopX

bluemoehre, 2023-10-24:
Hi @SwoopX , it was not on purpose to add a duplicate. I did search for an issue before, but due to naming I did not find it. 
Please tell me, what is the final solution to this issue if it is that old? It seems there is none.

In the code I can find manual [workarounds for different manufacturers](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/086ee3d32cc6488be92c33fc1740fd50ca24707d/simple_metering.cpp#L84) which makes the behavior even more unintentional.

Maybe you can point out the part of the code where the fields holding the divisors and multipliers are used to calc the correct final value for the API. Thanks.