dresden-elektronik/deconz-rest-plugin #7308: Device Request: Tuya PIR motion sensor

fsmaumke, 2023-10-20:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

809WZT

### Manufacturer

unknown (guess tuya)

### Model identifier

unknown

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Mimiix, 2023-10-20:
Hi.

Please fill the template completely. Make sure to add all clusters and them being read. the manufactuere name / model ID is in the basic cluster.

fsmaumke, 2023-10-21:
> Hi Mimiix,

please forgive me it's my 1st time doing things like that...

Please give me a hint - Cluster is just "blank".

So I have to change steeings (If yes, what needs 2 be changed) to her the correct View?

Greetings from Germany
Frank

Mimiix, 2023-10-21:
Hi Frank.

No worries.

You can find out here : https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

Kind regards,

fsmaumke, 2023-10-27:
Got it...

#the manufactuere name / model ID is in the basic cluster
=> Zigbee PIR _TZ3040_bb6xaihh / TS0202

[image]

fsmaumke, 2023-10-27:
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3040_bb6xaihh",
  "modelid": "TS0202",
  "product": "TS0202",
  "sleeper": false,
  "status": "Draft",
  "subdevices": []
}

fsmaumke, 2023-11-18:
I tought it's up to the Developer -Crew to implement this device...

Mimiix, 2023-11-18:
> I tought it's up to the Developer -Crew to implement this device...

It is, but unless anyone wants to help nothing happens. Nobody is obligated to help out.

@Smanar do you want to help out?

Smanar, 2023-11-18:
Sure it's just one more clone, cn try this DDF

```
{
   "schema":"devcap1.schema.json",
   "manufacturername": ["_TYZB01_tv3wxhcz","_TYZB01_hqbdru35","TZ3000_tiwq83wk","_TZ3000_ykwcwxmz","_TZ3000_hgu1dlak","_TZ3000_hktqahrq","_TZ3000_kmh5qpmb",
                        "_TZ3000_jmrgyl7o","_TZ3000_msl6wxk9", "_TZ3000_otvn3lne", "_TZ3000_6ygjfyll", "_TZ3040_6ygjfyll", "_TZ3000_mcxw5ehu","_TYZB01_ef5xlc9q",
                        "_TYZB01_vwqnz1sn","_TYZB01_2b8f6cio","_TZE200_bq5c8xfe","_TYZB01_dl7cejts","_TYZB01_qjqgmqxr","_TZ3000_mmtwjmaq","_TYZB01_zwvaj5wy",
                        "_TZ3000_bsvqrxru", "_TZ3000_mg4dy6z6", "_TZ3040_bb6xaihh"],
   "modelid":["TS0202", "TS0202", "TS0202", "TS0202", "TS0202","TS0202", "TS0202", "TS0202", "TS0202", "TS0202","TS0202", "TS0202", "TS0202", "TS0202",
             "TS0202", "TS0202", "TS0202", "TS0202", "TS0202","TS0202", "TS0202", "TS0202", "TS0202", "TS0202"],
   "product":"TS0202 Presence sensor",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint": {
           "profile": "0x0104",
           "device": "0x0402",
           "endpoint": "0x01",
           "in": [
             "0x0000",
             "0x0001",
             "0x0500"
           ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
              "name": "attr/swversion",                   
              "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
              "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
               "default":0
            },
            {
               "name":"config/duration"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
               "name":"state/presence",
               "awake":true,
               "parse":{
                  "fn":"ias:zonestatus",
                  "mask":"alarm1"
               }
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":3600,
               "max":14400,
               "change":"0x00000001"
            }
         ]
      }
   ]
}
```

fsmaumke, 2023-11-19:
OK, how can I help?  But my developer skills are at best at the lowest
level...

Smanar, 2023-11-19:
What is you OS ? to intall the DDF.
To make short, you just need to put a text file, renamed with the *.json extension and put the file in a "devices" folder. But some OS are not easier to use.

fsmaumke, 2023-11-19:
Hi, I use a raspi4 with the suggested OS: Raspbian

Greetings Frank

Smanar, 2023-11-19:
No docker ? Nice so will be the easier

`All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.`

So if you want to make a fast test, you can edit the file /usr/share/deCONZ/devices/tuya/ts0202_presence_sensor.json and add "_TZ3040_bb6xaihh" in manufacturename and another "TS0202" in modelid.

If you don't want to edit the deconz system file, you can create a complete new one in /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/device/tuya.

Then just restart deconz and include your device.

fsmaumke, 2023-11-26:
[image]
THX - seems 2 work ;-) 
BTW - I palced the [ddf](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7308#issuecomment-1817574760) in  /home//.local/share/dresden-elektronik/deCONZ/device/tuya

Smanar, 2023-11-26:
Thx, PR done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7385

closed 2023-11-28 by manup

mirml, 2023-12-16:
Will this also be included in Phoscon/deConz in Home Assistant?

Smanar, 2023-12-16:
In HA for sure, in Phoscon, I thin too, it have worked here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7308#issuecomment-1826768634