dresden-elektronik/deconz-rest-plugin #7309: Device Request: MHCOZY ZG-005-RF

ebaauw, 2023-10-20:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

MHCOZY 4-Channel Dry Contact Smart Wireless Zigbee Switch Relay Module

### Manufacturer

_TZ3000_u3oupgdy

### Model identifier

TS0004

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off
[image]

### Tuya Specific
[image]

Smanar, 2023-10-20:
I don't understand how the manufacture name/model id can be different on basic cluster and node information ?

Mimiix, 2023-10-20:
Erik and a Tuya device. Wauwie

ebaauw, 2023-11-17:
Ping

ebaauw, 2023-12-10:
Ping

closed 2024-01-09 by github-actions[bot]