dresden-elektronik/deconz-rest-plugin #7311: Lonsonho (Tuya) 4 gang: ZBWS04A / _TZ3000_uaa99arv / TS0044

rafalw1277, 2023-10-20:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Lonsonho (Tuya) 4 gang

### Manufacturer

_TZ3000_uaa99arv

### Model identifier

ZBWS04A

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2023-10-20:
Hello what is your deconz version ?
Because I m seing your device in the DDF > https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0044_4gang_remote.json

Nisbo, 2023-11-07:
I have the same device, running the latest BETA.
But my Manufacturer Name is different: **_TZ3000_wbfgbpxq** (2 devices with the same Name)
Can you please add the name to the list.

Smanar, 2023-11-07:
Have you tested the device ?
Because I don't found infoirmation about the _TZ3000_wbfgbpxq ? IDK what kind of device it's.

Nisbo, 2023-11-07:
Its the same than mentioned by [rafalw1277](https://github.com/rafalw1277) only with an different manufactur name
At the moment I can not add it, because the configurator is not able to pair it. So I can only test it in deconz.
Or what do you mean with "test it" ?

Where can I find the
`devices/tuya/_TZ3000_TS0044_4gang_remote.json`
file on my Raspberry Pi (with Conbee 2 Stick) ?
I tried `sudo find -name _TZ3000_TS0044_4gang_remote.json`
But with no results

I want to change
```
  "manufacturername": [
    "_TZ3000_ygvf9xzp",
    "_TZ3000_ee8nrt2l",
    "_TZ3000_dziaict4",
    "_TZ3000_mh9px7cq",
    "_TZ3000_b7bxojrg",
    "_TZ3000_ufhtxr59",
    "_TZ3000_abci1hiu",
    "_TZ3000_wkai4ga5",
    "_TZ3000_uaa99arv"
  ],
  "modelid": [
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044"
  ],
```

to

```
  "manufacturername": [
    "_TZ3000_ygvf9xzp",
    "_TZ3000_ee8nrt2l",
    "_TZ3000_dziaict4",
    "_TZ3000_mh9px7cq",
    "_TZ3000_b7bxojrg",
    "_TZ3000_ufhtxr59",
    "_TZ3000_abci1hiu",
    "_TZ3000_wkai4ga5",
    "_TZ3000_uaa99arv",
    "_TZ3000_wbfgbpxq"
  ],
  "modelid": [
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044",
    "TS0044"
  ],
```
to check, if I can pair the device.

Smanar, 2023-11-07:
Yep, it's exaclty that, and tuya make random devices, so not possible to be sure, they are same.
What is your OS ? Raspbian or a docker ?

On raspbian 

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

And after the modification, need to relaod deconz.

Nisbo, 2023-11-07:
I am using the Phoscon SDCARD Image for Raspberry PI 3 B+.
Thx for the headsup, I will give it a try after my afternoon shift or maybe tomorrow morning.

Smanar, 2023-11-07:
Oups, I forget something.
With the DDF your device will be supported but not working, you need to edit the xml file `/usr/share/deCONZ/zcl/general.xml`
And yur device to the list
```

        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_yw5tvzsk", "_TZ3000_czuyt8lz","_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_4upl1fcj", "_TZ3000_i3rjdrwu", "_TZ3000_w3c7ouru", "_TZ3000_abci1hiu","_TZ3000_wqcbzbae", "_TZ3000_pkeqinnt", "_TZ3000_uaa99arv"],
            "map": [
```

Nisbo, 2023-11-07:
@Smanar Are you sure `/usr/share/deCONZ/zcl/general.xml` is the correct file ? I am not able to find Tuya3gangMap in this mal. Additionally its not a JSON file, its XML

Smanar, 2023-11-07:
Lol, yeah, sorry, bad timing, so it's the json file, right.

Nisbo, 2023-11-08:
_TZ3000_wbfgbpxq  / TS004 is working

Not sure if it is a problem with the configuration or the description on Amazon (translation to German)
https://www.amazon.de/gp/product/B0BRV51VLX/ref=ask_ql_qh_dp_hza
It says 

> Vier benutzerdefinierte Schaltflächen, auf die Sie klicken, doppelklicken und lange drücken können, um die gewünschte Szene zu definieren

That means 4 buttons and you can 
- click, 
- double click and 
- long click 
but long click always enters the configuration / pairing mode of the remote.

I read something about this also in the reviews.

Also the battery (brand new one) shows only 28% (for both switches) but maybe this is a problem if IP-SYMCON. However, in deCONZ the battery is also indicated as Yellow.

At the moment, I am happy. Adding a remote, instead of using Alexa voice control, raises the WAF (wife acceptance factor) *g*

Smanar, 2023-11-08:
Perhaps with a shorter long press, but yes, it's possible tuya don't make new device, it just flood the market, with lot of unfinished devices using old devices.

For the battery It's a good question, there is so much DDF version, one for locked version (not the one you have used) and one for battery leak (you are not using it too)

You will see after some days, I can explain what to add the the DDF.

But if it work you can make a PR to add it, or I can make it if you want.

Nisbo, 2023-11-08:
Please add the PR for me. Thx :)

Smanar, 2023-11-08:
Done here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7341
Tell me if you have battery issue.

closed 2023-11-13 by manup