dresden-elektronik/deconz-rest-plugin #7312: Missing _TZ3000_p6ju8myv in DDF and Button Maps of Woox Smart Remote Control R7054

MNagler01, 2023-10-20:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

According to the website:
https://zigbee.blakadder.com/Woox_R7054.html
Woox Smart Remote Control R7054 with Zigbee ID: TS0215A, _TZ3000_p6ju8myv is supported by deCONZ.

The device can be paired but does not appear in the Phoscon App.

The reason seems to be that the Manufacturer: _TZ3000_p6ju8myv is missing in the DDF:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/woox/woox_r7054_remote.jso
and the Button Maps.

The Device has the following DDF Preview:
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_p6ju8myv",
  "modelid": "TS0215A",
  "product": "TS0215A",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0401",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0500",
          "0x0501"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/enrolled",
          "description": "State of IAS enrollment process."
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending",
          "description": "Pending tasks to configure the device."
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "description": "The last received button event."
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "description": "True when the device battery runs low."
        },
        {
          "name": "state/tampered",
          "description": "True when the device tampered alarm was triggered."
        }
      ]
    }
  ]
}

### Steps to reproduce the behavior

1. Go to Devices, Switches in the Phoscon app.
2. Press connect new switch.
3. Put the device into pairing mode.
4. The LED initially flashes blue.
5. The LED turns solid blue when connected.
6. But the device does not appear as a new switch in the Phoscon app.

### Expected behavior

Include the device:
Woox Smart Remote Control R7054 with Zigbee ID: TS0215A, _TZ3000_p6ju8myv
with all properties, buttons, etc. in deCONZ or its REST API.

### Environment

- Host system: Raspberry Pi
- Running method: Marthoc Docker container
- Firmware version: 26720700
- deCONZ version (not Home assistant Addon version!): 2.23.02
- Device: ConBee II
- Do you use an USB extension cable: yes -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?
a SSD

Smanar, 2023-10-20:
Right deconz miss this device, there is 3 clones of this devices, and deconz miss this one.
Can you make tries on your side to confirm it can work ?

You need to edit devices\woox\woox_r7054_remote.json
```
   "manufacturername":["_TZ3000_0zrccfgx", "_TZ3000_p6ju8myv"],
   "modelid":["TS0215A", "TS0215A"],
```

And add it in button_maps.json

```
        "TuyaKeyfobMap": {
            "vendor": "Woox",
            "doc": "4 button remotes from Woox and Nedis",
            "modelids": ["_TZ3000_0zrccfgx", "_TZ3000_fsiepnrh", "_TZ3000_p6ju8myv"],
            "map": [
                [1, "0x01", "IAS_ACE", "ARM", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm day/home zones only"],
                [1, "0x01", "IAS_ACE", "ARM", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Disarm"],
                [1, "0x01", "IAS_ACE", "EMERGENCY", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Emergency"],
                [1, "0x01", "IAS_ACE", "ARM", "3", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Arm all zones"]
            ]
        },
```

MNagler01, 2023-10-20:
I entered "_TZ3000_p6ju8myv" in the two files as you suggested and restarted the Docker container. I deleted the remote control using the VNC viewer and re-paired it using the Phoscon app. The device appeared after some time as a switch with four buttons in the Phoscon app called "Fernbedienung". I renamed it to "Sicherheitsfernbedienung 1".

As you can see from the [REST API.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13058897/REST.API.txt) file, there are only a few  entities for the device and the state of “lowbattery” is null.

I want to use the device in Home Assistant using the deCONZ integration. The device was created there without entities. When I press the buttons on the remote control, the events are displayed in the logbook. But there are no corresponding triggers that I could use for automations or scripts.

I tried to read the state of "buttonevent" as a sensor with RESTful API [rest.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13058996/rest.txt). The corresponding sensor received the state unknown.

What could be the reason?

Smanar, 2023-10-21:
Woaw, how you have done that so easily ? Changes are not persistent on dockers, you know as much docker ?

The state/buttonevent value is updated every time you use a button ? (a different one as it show the last one)

For HA perhaps this part will be usefull https://www.home-assistant.io/integrations/deconz/#remote-control-devices

MNagler01, 2023-10-21:
Thank you for the nice feedback. I'm still learning to use Docker.

I created a subdirectory "devices" in the directory that is connected to the container's /opt/deCONZ directory for permanent storage. I copied the file [button_maps.json](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13060913/button_maps.json) (contains only the basic informations and the modified button mapping) into this subdirectory.
I also created a subdirectory "woox" in the "devices" subdirectory into which I copied the file [woox_r7054_remote.json](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13060923/woox_r7054_remote.json).
Then I was able to restart the Docker container without losing this information.

Thank you also for pointing out the remote control section in the documentation for the deCONZ integration in HA. That helped me alot. I can now use the remote control as planned.

It would be nice if deCONZ or its REST API would report the battery state.

Smanar, 2023-10-21:
Ha you are using the officila docker image.
So yes creating/editing DDF is easier, but on my last check was not possible to edit the "button_maps.json
https://github.com/deconz-community/deconz-docker/issues/174

For the battery state, if the value exist on this device
[image]
IDK what it was not present in the DDF, probably not working. Can make try with adding

```
        {
          "name": "config/battery",
          "refresh.interval": 86400,
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          }
        },
```

On the sensor part and in the binding table

```
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
```

Can mimic for exemple devices\adeo\ldsenk08_door_vibration_sensor.json

MNagler01, 2023-10-21:
I made the suggested changes in the DDF. The battery state appeared as an entity in HA, but remained unchanged at 0. Apparently the value does not exist on this device.

The state of lowbattery also remains null and is not passed to HA as an entity. Any idea what this could be?

Smanar, 2023-10-22:
>The state of lowbattery also remains null and is not passed to HA as an entity. Any idea what this could be?

Yes, If I m right during the "device transfert" from deconz to HA, if a value is "null" it mean not supported for HA.
Can try
```
        {
          "name": "state/lowbattery",
          "default": false
        },
```

MNagler01, 2023-10-22:
Many thanks for your support. I set "refresh.interval" in "config/battery" to 3600 and "default" in "state/lowbattery" to false in the DDF.

The value of "battery" remains unchanged 0 when I reload the REST API information for the remote control in the Phoscon App after more than an hour. There is also no corresponding data among the events.  When I press a button the value also remains unchanged. I assume that the device does not provide this information. I will remove the corresponding entries in the DDF.

You're right, "lowbattery" is now displayed as "Low Battery" entity in HA. Unfortunately, I wasn't able to test whether this value changes to true when a low battery level occurs because I don't have a battery with this state.

I am attaching the modified DDF:
[woox_r7054_remote.json](https://github.com/dresden-elektronik/deconz-rest-plugin/files/13063315/woox_r7054_remote.json)

What happens next?

Smanar, 2023-10-22:
Depend of you ^^.
If all is done for you, you can submit a PR (for the support be official) I can do it if you want.

MNagler01, 2023-10-22:
Thank you for your offer. Please submit the PR for me.

closed 2023-11-06 by manup