dresden-elektronik/deconz-rest-plugin #7313: Issue with IKEA Fytur Blackout Blind

GoTTi74, 2023-10-20:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

For a few months, all my IKEA Fytur blinds show status "closed" even if open between 1-99% only when fully open, the status shows "open". I noticed that there were some changes in the DDF (devices/ikea/fyrtur_block-out_roller_blind.json) in the last few months for the "expression" of the "state/open".

When changing line 55
from:
`            "eval": "Item.val = Attr.val === 0"`
to:
`            "eval": "Item.val = Attr.val <100"` (as it is in the in the "tredansen_block-out_cellul_blind.json" and works in my integration) it acts as it is suppose to be.

Could someone review the current DDF?

### Steps to reproduce the behavior

In HomeAssistant:
1. Set the cover position to any open state between 1-99%
2. Go to "Developer Tools"
3. Search for the cover and check the state
4. Unless the cover is fully open, the state will say "closed"

In deCONZ:

1. Close deCONZ
2. Change DDF file as mentioned in the "bug description"
3. Start deCONZ
4. Reproduce steps in HA and check state. It will say "open" unless fully closed

### Expected behavior

When cover position is between 0 - 99% the state should show "open"
Only when cover is fully closed (100%) the state should show "closed"

### Environment

- Host system: Raspberry Pi
- Running method: Raspbian
- Firmware version: 0x26780700
- deCONZ version (not Home assistant Addon version!): 2.23.2
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? Yes! X857 mSATA SSD Shield expansion board via USB.

ebaauw, 2023-10-20:
Indeed, that's wrong.

GoTTi74, 2023-10-20:
One more thing I noticed although I've corrected the DDF, in Phoscon the state of the Fyrtur blinds isn't showing at all.
[image]

GoTTi74, 2023-10-20:
Also, it seems like the "TREDANSEN" still has the deprecated "state/on"?!
[image]

Mimiix, 2023-10-20:
> One more thing I noticed although I've corrected the DDF, in Phoscon the state of the Fyrtur blinds isn't showing at all. [image]

For phoscon, it needs to be reported on the forums :)

ebaauw, 2023-10-20:
> Also, it seems like the "TREDANSEN" still has the deprecated "state/on"?!

That's fixed in v2.24, currently still in beta.

closed 2023-11-06 by manup