dresden-elektronik/deconz-rest-plugin #7318: _TZ3000_yxqnffam - TS0203 - Tuya Smart Zigbee Door Window Contact Sensor ZigBee 3.0

Jupp2, 2023-10-25:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya Smart Zigbee Door Window Contact Sensor ZigBee 3.0

### Manufacturer

_TZ3000_yxqnffam

### Model identifier

TS0203

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Jupp2, 2023-10-25:
@Smanar
Hello Smanar, This is the Door Window Sensors I have. Only difference I see is manufacturer compared to:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5819 and https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5301#.
Could you add the clone manufacturer?
I made them somehow work at my home, by modifying the DDF.

Mimiix, 2023-10-25:
Please add all screens.

Mimiix, 2023-10-25:
> He have tested the device, so it's more an information than a device request. And it's certainly another clone, we can't make direclty a DDF update in this situation ?
> 
> He could have made a PR, so without capture.

I prefer to have all screens, regardless if its correct.

Jupp2, 2023-10-25:
I added the screens which are not fully empty. Hopefully this is "all" ?

Smanar, 2023-10-26:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7321

closed 2023-11-06 by manup