dresden-elektronik/deconz-rest-plugin #7319: TRADFRI bulb GU10 CWS 345lm

bluemoehre, 2023-10-26:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

TRADFRI LED 345lm

### Manufacturer

IKEA of Sweden

### Model identifier

TRADFRI bulb GU10 CWS 345lm

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off

[image]

### Level Control

[image]

### Color Control

[image]

ebaauw, 2023-11-17:
What's the _Image Type_ of this light?

bluemoehre, 2023-11-20:
Yes, it's `0x2802`. But if you just merge it into the other light I am afraid we cannot fill the field `product` correctly. That's another point why I wrote in my linked PR #7320 that I prefer a JSON for each device. But maybe I am wrong?

closed 2024-01-11 by github-actions[bot]