dresden-elektronik/deconz-rest-plugin #7322: Tuya ZTH05 Temperatur and Humidity Sensor with Display

weaslflink, 2023-10-27:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya ZTH05

### Manufacturer

Tuya

### Model identifier

TS0601 / _TZE204_upagmta9

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

bought this sensor from aliexpress
[Link to Aliexpress](https://www.aliexpress.com/item/1005005497473916.html?spm=a2g0o.order_list.order_list_main.22.350b5c5fJ7bHyI&gatewayAdapt=glo2en)

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2023-10-27:
Hello, all thoses devices are clone

`fingerprint: tuya.fingerprint('TS0601', ['_TZE200_yjjdcqsq', '_TZE200_9yapgbuv', '_TZE204_upagmta9', '_TZE200_cirvgep4']),`

So can make a try with this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_TS0601_humidity_temp.json

Just add your device in the list (need to add the manuyfacture name AND the model id even they are same)

weaslflink, 2023-10-27:
Hello,
I've modified the mentioned file as below.
Sensor is now recognized by Phoscon but the sensor has no values.

`  "manufacturername": ["_TZE200_nnrfa68v","_TZE200_znbl8dj5", "_TZE200_qoy0ekbd", "_TZE200_a8sdabtg", "_TZE200_whkgqxse","_TZE200_yjjdcqsq", "_TZE200_9yapgbuv", "_TZE200_utkemkbs","_TZE204_upagmta9"],
  "modelid": ["TS0601","TS0601", "TS0601", "TS0601", "TS0601", "TS0601", "TS0601", "TS0601", "TS0601"],
`

[image]

I am no expert in this but as far as I have understand the part of the code to get the values are here
`      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"`

what can I do on my side to find out the values ?
Is this something I can find [here ](https://developer.tuya.com/en/docs/connect-subdevices-to-gateways/zigbee-sensor?id=K9ik6zvmhrfh6)?

Smanar, 2023-10-27:
From what I m reading this device don't use classic cluster, but the tuya one. On your capture it don't have temperature or humidity cluster.

Edit:
Ok, this device need probably to be unlocked. Can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_yjjdcqsq","_TZE200_9yapgbuv", "_TZE204_upagmta9", "_TZE200_cirvgep4"],
  "modelid": ["TS0601", "TS0601", "TS0601", "TS0601"],
  "vendor": "Tuya",
  "product": "Temperature Humidity Sensor TS0601 locked version",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = (100 * Attr.val) + R.item('config/offset').val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
           "name":"config/tuya_unlock"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = (10 * Attr.val) + R.item('config/offset').val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

And you need to re-include it, it will make an unlock sequence at inclusion and every 24h.

weaslflink, 2023-10-27:
this also does not work.
however. I also noticed that the value on the display does not update. I guess I have received a defective sensor. I've asked the seller if he can give some information about display refresh/update cycle/behavior.

For me, this topic can be closed for now, as I seem to have defective hardware.

Smanar, 2023-10-28:
Can keep it open the time you receive your new device.
The one you already have is not yet in deconz, and you can have a new model again on your next one.

weaslflink, 2023-11-04:
I was in contact with the seller. He told me that the device has to be connected to **tuya** ZigBee gateway to work properly. The quick uncharged battery came from that the device was still looking for a tuya ZigBee gateway even if it is connected to the phoscon via Conbee II.

Smanar, 2023-11-04:
I m searching information about this device, it seem it need to make a TUYA_QUERY  after the device announce, but I don't found issue about this problem to have more description.

weaslflink, 2023-11-26:
> I m searching information about this device, it seem it need to make a TUYA_QUERY after the device announce, but I don't found issue about this problem to have more description.
> 
> Edit: Found some information [Koenkk/zigbee2mqtt#19137 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/19137#issuecomment-1785442408) Another shit from tuya.

Is there any chance to adapt the solution from z2m to deconz/phoscon ?

closed 2023-12-26 by github-actions[bot]