dresden-elektronik/deconz-rest-plugin #7324: PJ-1203A - Tuya Zigbee power meter with 2 clamps

aldoushx, 2023-10-29:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Electricity Energy Monitor

### Manufacturer

_TZE204_81yrt3lo

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]
[image]

### Endpoints and clusters

[image]
[image]

### Basic

[image]
[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2023-10-29:
You miss the node picture with the cluster list.
Can start with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE204_81yrt3lo", "_TZE200_rks0sgb7"],
  "modelid": ["TS0601", "TS0601"],
  "product": "PJ-1203A - Zigbee power meter with 2 clamps",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 106, "eval": "Item.val = Attr.val / 100;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
[... 180 lines cut ...]
          "name": "state/lastupdated"
        },
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 114, "eval": "Item.val = Attr.val / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid":112, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 105, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

aldoushx, 2023-10-29:
Thank you for your answer, I updated the 1st message with the node picture.

For an unknown reason, I cannot edit/save the ddf in the editor with the ddf provided above : I open the json in the ddf editor window of the device, the data is well  displayed in the ddf editor, but when i save it, close the window then go back in the editor, the ddf is back to the initial data...
I tried after a hot reload, and after deleting/reinclude the node, but i get the same result every time, the ddf is not saved.

Any idea what it could be ? That is the first time I see this behaviour with a ddf...

Smanar, 2023-10-29:
You can't use the editor.
What is your OS ? You need to create a text file called what_you_want.json in a "devices"  folder, but it will depend of your OS.

aldoushx, 2023-10-29:
I know the procedure, but the ddf.json file does not stay loaded in the editor after I opened it and saved (unlike the other ddf files I used for other devices).

Smanar, 2023-10-29:
Don't use the editor, I don't think it's able to manage tuya device, better to use a text editor.
Close deconz, put the file in one of the 2 "devices" folder, and reload deconz.

Smanar, 2023-10-30:
You have zero values or bad one ? Perhaps the /100 is too much.
You haven't the GUI to check logs ? with deconz/help/debug veiw with flag "DDF"

aldoushx, 2023-10-30:
The consumption values were ridiculously low in Home Assistant, so I assumed they were not returned. 
I then changed the /100 by *10 in the ddf and now the values are the right ones (in Wh in deconz debug).
I can also see the current and voltage values in the deconz debug window, but these values are not returned as sensors in HA.

Smanar, 2023-10-30:
Can you take a look direclty in the api ?
Using phoscon/help/API Information/sensors.

aldoushx, 2023-10-31:
It the API, the voltage and current values are displayed (curently at 0)

                    
```
{
    "config": {
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "etag": "11d7731910045ed5a11b959d8704db9b",
    "lastannounced": null,
    "lastseen": "2023-10-31T21:06Z",
    "manufacturername": "_TZE204_81yrt3lo",
    "modelid": "TS0601",
    "name": "Power 113",
    "state": {
        "current": 0,
        "lastupdated": "2023-10-31T21:06:59.519",
        "power": 0,
        "voltage": 236
    },
    "type": "ZHAPower",
    "uniqueid": "a4:c1:38:1e:ff:ed:10:fd-01-0b04"
}
```

Smanar, 2023-11-01:
Ok, so you know the method to "refresh" a device in HA ? for exemple delete it in HA and not deconz, then relaod it ?

If I m right HA create the entry the first time it see the device, so if the state/voltage was missing at this moment, it will stay missing, even it will apear later.

Because all seem fine for me, no reason to don't have the voltage.

Pirionfr, 2024-05-20:
up

Smanar, 2024-05-20:
Hello, can you try the DDF too ?
Now with nthe DDF bundle this step is easier, you just need the 0.27 version.

mc7547, 2024-09-24:
Good morning 
I have the ComBee II Gataway version 2.28.1
I installed a "PJ-1203A - Tuya Zigbee power meter with 2 terminals" and on first use it didn't report anything to me.
Following the suggestions above I modified the multiply and divide values ​​of the parameters. Now I have

{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE204_81yrt3lo", "_TZE200_rks0sgb7"],
  "modelid": ["TS0601", "TS0601"],
  "product": "PJ-1203A - Zigbee power meter with 2 clamps",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 106, "eval": "Item.val = Attr.val * 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
[... 183 lines cut ...]
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 114, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid":112, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 105, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}

even with the load deactivated it still gives me consumption.
How can I intervene?

mc7547, 2024-09-24:
Thanks for any help
Rob

Smanar, 2024-09-24:
Hello, so the device is working for you, you have report, but wrong "scale" ?

mc7547, 2024-09-24:
HI
What do you mean by report? Maybe the deConz API JSON? Yes, I have that.
I tried different multiplying and dividing factors and the best I got is the one I attached in the previous POST.

Thank you
I'm waiting

roby

Smanar, 2024-09-25:
Using multiplying and dividing factors is not enought according what I m seing, you need too chancge the dpid
eg for the first one the code need to be

```
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": **101**, "eval": "Item.val = Attr.val / 100;" },
          "read": {"fn": "none"},
          "default": 0
        },
```

Smanar, 2024-09-26:
It seem not so bad, can you show a "bad value sample" ? (the value, you, the one you need to have and the field)

For information on the GUI, in help/debug view, you can enable the flag "DDF" (and "info" + "info_l2") with them you will see the raw value and the "magic" done by the DDF.

mc7547, 2024-09-26:
Thanks for all the suggestions and I think I managed to create a satisfactory DDF (which I attach below).
I have two problems:
1) I don't know how to put intervals and or timeouts in the DDF to avoid being flooded with messages (via websoket) 
2) I would like someone to test the DDF and validate my work

Thank you

Rob

{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE204_81yrt3lo", "_TZE200_rks0sgb7"],
  "modelid": ["TS0601", "TS0601"],
  "product": "PJ-1203A - Zigbee power meter with 2 clamps",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 106, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
[... 184 lines cut ...]
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 114, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid":112, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 105, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}

mc7547, 2024-09-26:
HI
I actually have a third problem. 
In fact, now I use both channels to measure the data during the consumption phase of the devices but how should I modify the DDF to also control the data produced by the photovoltaic?

Thanks for any help

roby

Smanar, 2024-09-28:
1- Not possible with tuya
2 - If it work for you, no problem for me, but to make the PR better to make a new DDF than "mixing" the 2 devices as they are using different dpid > different DDF.
3- You have only 2 torus, so you have 2 returns, you mean positive or negative ?

mc7547, 2024-09-28:
What I mean by point 3 is that I would like to use one toroid to measure the data relating to my consumption and the other toroid to measure the production data of my photovoltaic system. I would like to know if it is possible to manage everything with a single DDF and how to do it or if I need to act in another way. But which one?
Thank you
Rob
P.s. I'm doing more tests and other changes to the DDF. In fact there are some strange behaviors. Let's hope we get to the bottom of it

Smanar, 2024-09-28:
But it's electrical question ?
It not just depend where you put the toroid ? production or consumption is the same thing, except the sign perhaps ?

Smanar, 2024-09-28:
But it's electrical question ?
It not just depend where you put the toroid ? production or consumption is the same thing, except the sign perhaps ?

There is a state/production usable on DDF but on your situation, I don't see the utlity ?

mc7547, 2024-09-29:
HI
I don't know if it's the same thing and that the sign needs to be changed but 
the specs also talk about "energy_produced_a" and "energy_produced_b" (besides "energy_a" and "energy_b" work well) so I think there is a difference between the energy I take from the energy I produce. Furthermore, regarding the data output interval, I found the "update_frequency" attribute but I don't know how to insert it into the DDF. Now it produces too much data and much of it is redundant.
As far as usefulness is concerned, in my situation it cannot be used at the moment because I don't know how to configure the DDF to manage both situations. That is, read both consumption and production data with the same DDF

Thanks for the help provided.
Rob

Smanar, 2024-09-29:
Ha yes, haven't see them

```
                [102, 'energy_produced_a', tuya.valueConverter.divideBy100],
                [104, 'energy_produced_b', tuya.valueConverter.divideBy100],
                [2, 'energy_produced', tuya.valueConverter.divideBy100],
```

For information, you can use a "state/production" item on your DDF (on the same entry than for the state/consumption" for exemple), but I don't see the difference with the consumption, as it will be the same toroid ...

Where have you see this attribute "update_frequency" ?

mc7547, 2024-09-29:
HI
I'll try to do as you say. As for the "update_frequency" attribute, I looked into it further and I think I misunderstood its meaning. It should be the updating of the frequency of the current and not the updating of the frequency of sending the data. But he carried out further checks. For now, what I have managed to obtain is enough for me and I intend to delve deeper

Thanks for the help

Rob

carryonrewardless, 2025-12-14:
I understand that as a concept but with deConz Addon being a container there are limited options for how to save the file. When you go into the docker environment and access the /usr/share/deCONZ/devices folder there is no file editor available e.g. nano or vi.  The only option is to use wget to get the file from external url storage but then the file doesn't survive a  deConz re-start. There is apparently a .local/.... /devices folder but have not been able to find that.
I have also placed the file in /root/homeassistant/devices folder but it is not picked up by the Addon.
Any help would be much appreciated.

Smanar, 2025-12-14:
If you have docker.
- You can use specific command for docker https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/20
- You can use https://deconz-community.github.io/ddf-tools/#/auto-bundler to make a DDB archive and use it with phoscon.

Smanar, 2025-12-15:
BTW if the DDF is working,  someone want to make a PR to submit the device officialy ?

closed 2026-01-07 by manup