dresden-elektronik/deconz-rest-plugin #7326: deonz-gui crashes on clicking any dropdown menu gui element.

opechunka, 2023-11-01:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

deconz-gui crashes on activating any dropdown menu. 

### Steps to reproduce the behavior

Start deconz-gui. Try to activeate any dropdown menu gui element. 
IAZone Cluster -> Zone Enroll response -> Enroll response code. 
By default "Success" is there. Try to click on it. The application crashes immediately and disappeare form the desktop. 

### Expected behavior

No crash while selectin dropdown menu

### Environment

- Host system: Raspberry Pi "Raspbian GNU/Linux 11 (bullseye)
- Running method: (Raspbian )
- Firmware version: (26720700)
- deCONZ version 2.23.00
- Device: ConBee II
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system? no

opechunka, 2023-11-20:
Well, when the deconz gui is started via command line, it doesn't crash anymore. Only when start from Raspbian Start menu.

Smanar, 2023-11-20:
It's not corrected on last version ?

opechunka, 2023-11-20:
> It's not corrected on last version ? (2.24.2)

After updateing to the latest version (or probably due to other changes/updates) deconz menu item dissapered from Raspbian Start menu. So i cannot test it.

SwoopX, 2023-12-11:
Resolved with v2.24.2

closed 2023-12-11 by SwoopX

opechunka, 2024-05-02:
Still crashes 2.26.3