dresden-elektronik/deconz-rest-plugin #7327: Support for LYWSD03MMC Xiaomi Sensor

vdiogo, 2023-11-01:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

LYWSD03MMC

### Manufacturer

Xiaomi

### Model identifier

LYWSD03MMC

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]
[image]
[image]
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

olleelfsborg, 2023-11-10:
Anything new on this?

molnarti, 2023-11-11:
how have you been able to pair the device to show up its data in deconz? mine is not pairing, no matter what i do (yes, i did flash it with the ZB firmware)

vdiogo, 2023-11-13:
> how have you been able to pair the device to show up its data in deconz? mine is not pairing, no matter what i do (yes, i did flash it with the ZB firmware)

I opened the Phoscon app, added a sensor and pressed the button in the device, and waited for it to show in the deconz network. Of course it never added the sensor in Phoscon, was just to force the conbee stick to scan for new devices

Mimiix, 2023-11-13:
> how have you been able to pair the device to show up its data in deconz? mine is not pairing, no matter what i do (yes, i did flash it with the ZB firmware)

It won't show in phoscon. Just deconz.

molnarti, 2023-11-13:
OK, so mine is paired too (was expecting to see something in phoscon and did not check deconz). Hope this gets supported soon, would be a game changer in my smart home setup

ostgardh, 2023-11-13:
I have add LYWSD03MMC in DDF editor, and now I can se LYWSD03MMC in phoscon

ostgardh, 2023-11-16:
But I get no update from LYWSD03MMC.
Anyone know how to add temp report when the temperature changes?

DrBlokmeister, 2023-11-20:
> I have add LYWSD03MMC in DDF editor, and now I can se LYWSD03MMC in phoscon. similarly as in this video.. https://www.youtube.com/watch?v=rgTQpOVg8w0

Could you share the .json file? As far as I understand, I can copy/paste that in my DDF editor, right?

neutrino85, 2023-11-20:
Hello,
Here's mine :
`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Xiaomi",
  "modelid": "LYWSD03MMC",
  "product": "LYWSD03MMC",
  "sleeper": true,
  "status": "Silver",
  "path": "/devices/lywsd03mmc.json",
  "subdevices": [
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 3600,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 7255,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true,
          "refresh.interval": 60,
          "default": 0
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001"
    }
  ]
}`

ostgardh, 2023-11-20:
This is my.
But I still have problem that is not report new temp.
I have to press update in deconz ..

File
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Xiaomi",
  "modelid": "LYWSD03MMC",
  "product": "LYWSD03MMC",
  "sleeper": false,
  "status": "Silver",
  "path": "/devices/lywsd03mmc.json",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "description": "The current device battery level in range
0&ndash;100 %.",
          "refresh.interval": 3600
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

btk999, 2023-11-22:
I also cant get it work, the device is shown in deCONZ but not in Phoscon or Rest API. Neither with my own DDF nor the DDFs posted above. Also after every deCONZ restart i have to reload the DDF...

DDF-God @Smanar  can you help us out here? :-)

Smanar, 2023-11-22:
Hello, can you share the node picture ? the one with all cluster and endpoint displayed ? (to be sure all is working on endpoint 0x01.
Else on your DDF you have set the bind, but forget to set the reporting

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "Xiaomi",
    "modelid": "LYWSD03MMC",
    "vendor": "Xiaomi",
    "product": "LYWSD03MMC",
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_HUMIDITY_SENSOR",
            "restapi": "/sensors",
            "uuid": [
                "$address.ext",
                "0x01",
                "0x0405"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "config/battery",
                    "refresh.interval": 3600,
                    "read": {
                        "at": "0x0021",
                        "cl": "0x0001",
                        "ep": 1,
                        "fn": "zcl"
                    },
                    "parse": {
                        "at": "0x0021",
                        "cl": "0x0001",
                        "ep": 1,
                        "eval": "Item.val = Attr.val / 2",
                        "fn": "zcl"
                    },
                    "default": 0
                },
                {
                    "name": "config/offset",
                    "default": 0
                },
                {
                    "name": "config/on"
                },
                {
                    "name": "config/reachable"
                },
                {
                    "name": "state/humidity"
                },
                {
                    "name": "state/lastupdated"
                }
            ]
        },
        {
            "type": "$TYPE_TEMPERATURE_SENSOR",
            "restapi": "/sensors",
            "uuid": [
                "$address.ext",
                "0x01",
                "0x0402"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "config/battery",
                    "refresh.interval": 7255,
                    "read": {
                        "at": "0x0021",
                        "cl": "0x0001",
                        "ep": 1,
                        "fn": "zcl"
                    },
                    "parse": {
                        "at": "0x0021",
                        "cl": "0x0001",
                        "ep": 1,
                        "eval": "Item.val = Attr.val / 2",
                        "fn": "zcl"
                    },
                    "default": 0
                },
                {
                    "name": "config/offset",
                    "default": 0
                },
                {
                    "name": "config/on"
                },
                {
                    "name": "config/reachable"
                },
                {
                    "name": "state/lastupdated"
                },
                {
                    "name": "state/temperature"
                }
            ]
        }
    ],
    "bindings": [
        {
            "bind": "unicast",
            "src.ep": 1,
            "cl": "0x0001",
            "report": [
                {
                    "at": "0x0021",
                    "dt": "0x20",
                    "min": 3200,
                    "max": 3600,
                    "change": "0x00000002"
                }
            ]
        },
        {
            "bind": "unicast",
            "src.ep": 1,
            "cl": "0x0402",
            "report": [
                {
                    "at": "0x0000",
                    "dt": "0x29",
                    "min": 10,
                    "max": 300,
                    "change": "0x0000000A"
                }
            ]
        },
        {
            "bind": "unicast",
            "src.ep": 1,
            "cl": "0x0405",
            "report": [
                {
                    "at": "0x0000",
                    "dt": "0x21",
                    "min": 10,
                    "max": 300,
                    "change": "0x00000032"
                }
            ]
        }
    ]
}
```

btk999, 2023-11-22:
> Hello, can you share the node picture ? the one with all cluster and endpoint displayed ? (to be sure all is working on endpoint 0x01. Else on your DDF you have set the bind, but forget to set the reporting
> 

Thanks! Seems that your DDF works - the sensor gets updated recently.
Here some screenshots - i hope this is what you asked for..?
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2023-11-22:
Yep, But if it works, it's fine.
You can make a PR to submit the device ? or do you want I make it ?

btk999, 2023-11-22:
> Yep, But if it works, it's fine. You can make a PR to submit the device ? or do you want I make it ?

to be honest i dont know how to do it. It would be great if you could do that :)

ostgardh, 2023-11-22:
But have you get sensor to report when temperature changes?

btk999, 2023-11-22:
> But have you get sensor to report when temperature changes?

yes, with the DDF from @Smanar this works.

ostgardh, 2023-11-22:
Ok. I have to test your ddf 

Thanks

GiacomoFilzi, 2023-11-23:
Looking very much forward to this release -awesome!

molnarti, 2023-11-23:
i can't make the DDF "stick". I click on the entity, select edit DDF, load the saved json file - it seems everything went ok. I click on save, then hot reload, but nothing changes, the device is not visible in Phoscon. After restarting phoscon/deconz the DDF is gone and the node is empty again

Smanar, 2023-11-23:
@molnarti what is your OS ? You areunder docker ?

PR done here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7373

molnarti, 2023-11-23:
> @molnarti what is your OS ? You are under docker ?
 yes, docker. i have placed the DDF in a bind mount, and it can load without apparent issues, but seems its not applied

Smanar, 2023-11-24:
The official deconz docker or the HA one ?
For the official one, DDF need to be in `opt/deCONZ/devices/`. Are you sure about your model, DDF are used according modelID+ManufactureName.

It's a custom device, you can have different one, actually the DDf check for

```
manufacturername : Xiaomi
modelid : LYWSD03MMC
```

molnarti, 2023-11-24:
yes, that is the location where my DDF is placed. 

how can i confirm my device is really the same? are there other Xiaomi BT sensors that can be flashed to zigbee?
[image]

btk999, 2023-11-25:
@molnarti  did you try to set deCONZ to Hybrid Mode? 
[image]

Smanar, 2023-11-25:
>how can i confirm my device is really the same? are there other Xiaomi BT sensors that can be flashed to zigbee?

The device is lazy, so you need to awake it when asking attribute.
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

Model id and manufacture Name are in the cluster 0x0000 "Basic"

closed 2023-11-29 by manup

kszabi777, 2023-12-01:
Thanks for sharing this, lifesaver.

Could you please explain to me what are the reporting values mean ? (at,dt,min,max,change) 
With the current settings how often the sensor reports the actueal temp/humidity ?

Smanar, 2023-12-01:
yep.

```
                {
                    "at": "0x0000",
                    "dt": "0x21",
                    "min": 10,
                    "max": 300,
                    "change": "0x00000032"
                }
```

According thoses setting you need to have a report at least every 300s or if the humidity increase/decrease 0.5°% (0x32 = 50, humidity is 4980 on your capture so 49.80 %)