dresden-elektronik/deconz-rest-plugin #7330: Incorrect handling of voltage/current/power in DDF for Namron panel heater

nekronos, 2023-11-01:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

DDF for [Namron panel heater](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/086ee3d32cc6488be92c33fc1740fd50ca24707d/devices/namron/5401395_heater.json) does not adjust the numbers for voltage, current and power. For example, power reports 10000w when in use. The heater is only 1000w.

### Steps to reproduce the behavior

See expected behavior and screenshot.

### Expected behavior

The device has attributes that reports a divisor and a multiplier that should be used to compute the actual value for voltage, current and power. The DDF needs to take this into account. See screenshot.

### Screenshots

[image]

### Environment

- Host system: Raspberry Pi 2
- Running method: Raspbian
- Firmware version: 0x26690700
- deCONZ version v2.23.2
- Device: RaspBee II

nekronos, 2023-11-01:
If there are any docs or good examples for how to write the DDF for this, please let me know and I'll be happy to look into this issue

BabaIsYou, 2023-11-03:
May be could you be inspired b the way these values are calculated for this thermostat : https://github.com/dresden-elektronik/deconz-rest-plugin/blob/086ee3d32cc6488be92c33fc1740fd50ca24707d/devices/ouellet/OTH4000-ZB.json

Smanar, 2023-11-03:
What are the value you need to correct ? can use code like this in the DDF

```
        {
          "name": "state/voltage",
          "refresh.interval": 360,
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Math.round(Attr.val / 100);",
            "fn": "zcl:attr"
          }
        }
```

Here the value is divided by 100

nekronos, 2023-11-27:
> As there has not been any response in 21 days, this issue has been automatically marked as stale. **At OP: Please either close this issue or keep it active** It will be closed in 7 days if no further activity occurs.

Got it working, will submit a PR. Just busy so have not found the time yet

closed 2023-12-27 by github-actions[bot]