dresden-elektronik/deconz-rest-plugin #7332: Installing 2.24 over 2.22 kills all config and does not see Conbee device - must restore from backup

Ltek, 2023-11-02:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I've been on 1.22 for a while and decided to install 2.24 and I tried 2.24.1 ... They both killed my install. I had uninstall the apps and restore from backup.

### Steps to reproduce the behavior

Have running / working 1.22 
install  2.24 or newer
see the install not work
restore 2.22 from backup to get it working

### Expected behavior

all other versions i've installed for years saved the config from the prior version

### Screenshots

[image]

### Environment

- Win11 PC
- install in root of C drive  (C:\deConz)
- Firmware version: (26xxyy00)
- 2.22 works
- ConBee I
- Yes, always haveI
- no

Rolf56, 2023-11-18:
I have the same issue in my installation!

[image]

The connected device is a ConBee I and not a ConBee III as shown in the screen

[image]

**Environment**

    Win10 PC
    install in C:\Program Files (x86)\deCONZ
    Firmware version: (0x26400500)
    2.23.1 works
    ConBee I

apper13, 2023-11-18:
I have the same issue after updating from 2.22.2 to 2.24.1. Conbee III shown in the dropdown list (instead of the connected Conbee I) and it is not connecting. Fortunately downgrading to 2.22.2 worked fine for me.

stoneubi, 2023-11-21:
Same here. After performing an update via apt-get update && upgrade on my virutal machine running dietpi on a proxmox server, deconz app was removed. Actually 2.23.02 is installed. After the metioned update, the whole deconz installation has crashed. Thx good that I have a proxmox server and that I do a full backup before upgrading.

stoneubi, 2023-11-21:
OK:1 http://phoscon.de/apt/deconz bullseye InRelease
OK:2 https://deb.debian.org/debian bullseye InRelease
Holen:3 https://deb.debian.org/debian bullseye-updates InRelease [44,1 kB]
Holen:4 https://deb.debian.org/debian-security bullseye-security InRelease [48,4 kB]
Holen:5 https://deb.debian.org/debian bullseye-backports InRelease [49,0 kB]
Holen:6 https://deb.debian.org/debian-security bullseye-security/main amd64 Packages [258 kB]
Holen:7 https://deb.debian.org/debian bullseye-backports/main amd64 Packages.diff/Index [63,3 kB]
Holen:8 https://deb.debian.org/debian bullseye-backports/main amd64 Packages T-2023-11-20-1403.46-F-2023-11-20-1403.46.pdiff [1.243 B]
Holen:8 https://deb.debian.org/debian bullseye-backports/main amd64 Packages T-2023-11-20-1403.46-F-2023-11-20-1403.46.pdiff [1.243 B]
Es wurden 464 kB in 1 s geholt (324 kB/s).
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Aktualisierung für 1 Paket verfügbar. Führen Sie »apt list --upgradable« aus, um es anzuzeigen.
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Paketaktualisierung (Upgrade) wird berechnet… Fertig
Die folgenden Pakete werden aktualisiert (Upgrade):
  deconz
1 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen 13,6 MB an Archiven heruntergeladen werden.
Nach dieser Operation werden 4.313 kB Plattenplatz freigegeben.
Holen:1 http://phoscon.de/apt/deconz bullseye/main amd64 deconz amd64 2.24.2 [13,6 MB]
Es wurden 13,6 MB in 3 s geholt (4.086 kB/s).
debconf: delaying package configuration, since apt-utils is not installed
(Lese Datenbank ... 72348 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../deconz_2.24.2_amd64.deb ...
Entpacken von deconz (2.24.2) über (2.23.02-debian-buster-stable) ...
deconz (2.24.2) wird eingerichtet ...
Trigger für mailcap (3.69) werden verarbeitet ...
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig

stoneubi, 2023-11-21:
After this update the desktop application is not available any more: /usr/share/applications/deCONZ.desktop

Ltek, 2023-11-22:
With these showstopper bugs and the lack of Tuya support I've decided to migrate (over time) to Z2M which supports all Tuya devices without hassle and I've only encountered minor bugs.

I'm not going to touch deconz and stay on the older version until all my devices are migrated.

MarioMarti77, 2023-12-04:
I had the same issue, try to run deconz with  --baudrate=38400

Rolf56, 2023-12-04:
baudrate=38400 does not help. Still the same issue!
I changed this in Windows device manager. Is there a other place to change?

Mimiix, 2023-12-04:
As the OP mentioned he has left the platform, there is no need to keep this open. @Rolf56  Please ask your question on the forums, as it is not compliant to #5113.

Closing and locking.

closed 2023-12-04 by Mimiix