dresden-elektronik/deconz-rest-plugin #7333: [New device support]: _TZE200_rxntag7i TS0601 / Tuya - Avatto TRV16

derdon999, 2023-11-02:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

TRV16

### Manufacturer

Tuya

### Model identifier

_TZE200

### Device type to add

None

### Node info

As I cannot connect to this device via deconz nor phoscon, I can only link the information given in the zigbee2mqtt thread: https://github.com/Koenkk/zigbee2mqtt/issues/18872

### Endpoints and clusters

{"id":51,"type":"EndDevice","ieeeAddr":"0xa4c138d4a8fef677","nwkAddr":10287,"manufId":4417,"manufName":"_TZE200_rxntag7i","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_rxntag7i","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1693927818317,"defaultSendRequestWhen":"immediate"}

### Basic

the device can be found here: https://www.aliexpress.com/item/1005005850177778.html

### Further relevant clusters

I really don´t know how to add this TVR16 to my Conbee II instance on Raspi with deconz. The devices are not linking/connecting and therefiore I don´t see them in phoscon, REST api or openHAB. Is it possible you give me a hint?

ConBee II
Gateway Version 2.24.1
Firmware 26780700
deconz-2.24.1-qt5.deb
openHAB 3.4.5

Thanks a lot!

Mimiix, 2023-11-02:
Please add all screens.

derdon999, 2023-11-03:
> Please add all screens. You need to check deconz.

Thanks for reminding me. As I already mentioned I don't see the device in deconz on my raspi nor on windows app. There are some unknown devices popping up but basic cluster or node info is empty or just showing 0x000. No name or anything given there. Did I miss something or where else can I check?
[image]
[image]

Mimiix, 2023-11-03:
Hi,

Thats Phoscon, not deCONZ.

derdon999, 2023-11-08:
Sorry for the late reply, I was not home. Is this enough?
[image]
[image]

Smanar, 2023-11-09:
All thoses devices are clone

```
        fingerprint: tuya.fingerprint('TS0601', [
            '_TZE200_bvu2wnxz', /* model: 'ME167', vendor: 'AVATTO' */
            '_TZE200_6rdj8dzm', /* model: 'ME167', vendor: 'AVATTO' */
            '_TZE200_gd4rvykv', // Sanico
            '_TZE200_p3dbf6qs', /* model: 'ME168', vendor: 'AVATTO' */
            '_TZE200_rxntag7i', /* model: 'ME168', vendor: 'AVATTO' */
        ]),
```

Can make a try with `devices\tuya\_TZE200_TS0601_trv.json` made for 

```
  "manufacturername": ["_TZE200_bvu2wnxz", "_TZE200_6rdj8dzm", "_TZE200_gd4rvykv"],
  "modelid": ["TS0601", "TS0601", "TS0601"],
```

derdon999, 2023-11-09:
> devices\tuya\_TZE200_TS0601_trv.json

After reading the docs and more information on Github I was able to add the new device in the corresponding DDF from here https://dresden-elektronik.github.io/deconz-rest-doc/devices/tuya/_TZE200_TS0601_trv/ and copy it to the folder usr/share/deCONZ/devices/tuya on my Raspberry. After restarting I was able to see the thermostats and control them. But not everything seems to work. 
Temperature range on device: 5-35°C. In Phoscon range 5-30°C.
Offset can´t be adjusted. Time schedule not availabe, but can be turned on/off on device. Safety lock with no action.
[image]

And in openhab I see those channels:
[image]

Smanar, 2023-11-10:
>fter reading the docs and more information on Github I was able to add the new device in the corresponding DDF from here https://dresden-elektronik.github.io/deconz-rest-doc/devices/tuya/_TZE200_TS0601_trv/ and copy it to the folder usr/share/deCONZ/devices/tuya on my Raspberry

You haven't it natively ? what is your deconz version ?
The original file is here https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_TS0601_trv.json

But first time I m seeing a support for TRV in phoscon !

What is your offset ? a temperature offset or an "hysteresis" for regulation ? Both are missing in DDF, but not the same thing to add.

>Temperature range on device: 5-35°C. In Phoscon range 5-30°C.

Heeeee, here there is nothing about range in DDF, can be a phoscon choice.

You can use config/mode to change mode, but IDK how to do in phoscon.

derdon999, 2023-11-11:
> You haven't it natively ? what is your deconz version ?
> The original file is here https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_TS0601_trv.json

See details to my system in the first post... deconz-2.24.1-qt5.deb.
The link you posted is the one I posted too ;)
Natevely it was not working, because details have been missing. So I added the name _TZE200_rxntag7i and details like this:
`{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZE200_bvu2wnxz",
    "_TZE200_6rdj8dzm",
    "_TZE200_gd4rvykv",
    "_TZE200_rxntag7i"
  ],
  "modelid": [
    "TS0601",
    "TS0601",
    "TS0601",
    "TS0601"
  ],`

Now I was able to add the device in phoscon and openhab and control it. Right now I´m just able to set the temperature directly at the device and openhab or set it to HEAT or OFF from openhab. Offset or hysteresis I can set but don´t see any changes.

Smanar, 2023-11-11:
>I´m just able to set the temperature directly at the device and openhab or set it to HEAT or OFF from openhab

So it work fine with openHAB ? it's the principal, we haven't impact on Phoscon, and it's the first time I see a support for them inside.

>Offset or hysteresis I can set but don´t see any changes.

But on wich one field are you making changes ? I don't see something in the DDF.
From z2m it seem it's just an offset

>local_temperature_calibration: Offset to be used in the local_temperature. To control publish a message to topic 

So can add in the DDF
```
        {
          "name": "config/offset",
          "default": 0
        },
```

And change temperature field for

```
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 5, "eval": "Item.val = Attr.val * 10 + R.item('config/offset').val;"},
          "read": {"fn": "none"}
        }
```

derdon999, 2023-11-12:
> So will be able to use config/offset.

Thanks. I added this, replaced the old DDF, restarted my raspberry but don´t see any changes in Phoscon nor more details. If it´s working from openhab as sendcommand and taking any effects, I can´t say right now.

Smanar, 2023-11-12:
Yeah will be probably not supported in Phoscon, but probably more chance on openhab.
The change need to be visible realy fast, can try an offset of 500 direclty (for 5 °C), you can too if it have worked direclty on the API, IDK how to do with openHAB but you can use phoson/help/API Information/sensors/device_id

derdon999, 2023-11-26:
Offset is working, Thanks!
Still missing in Phoscon and openhab:
Temperature range on device: 5-35°C. In Phoscon/openhab range 5-30°C.
Time schedule not availabe, but can be turned on/off on device and as Auto in openhab. But were to configure the details? Or is this just another state like HEAT or OFF as it is show with it together as options. 
Safety lock with no action.
FYI: Latest update of deconz deleted my config.

Smanar, 2023-11-27:
>FYI: Latest update of deconz deleted my config.

Yeah, you have 2 "devices" folder, one is used by deconz, reset on every update, and the second one in the "user" folder, untouched.

> 
All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

>Safety lock with no action.

Can you take a look on logs (deconz/help/debug view with "info" and "info_l2") when you use the API and when you set it manualy to compare request ?

>Temperature range on device: 5-35°C. In Phoscon/openhab range 5-30°C.

You have an error message if you use out of limit values ? Because I haven't set limit on my side.

For the shedule I m seeing there is tuya request by day to set them, but idk how to do with existing command on DDF ....

Smanar, 2023-12-01:
To get logs you need the GUI, harder with command line https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-command-line-parameters#available-debug-parameters

After the theory is simple, just make an action phisicaly on the device, get the request on logs, and you will just to mimic it with DDF after.

Smanar, 2023-12-05:
Just enable flag "info" and "info_l2" , if you are using command line `--dbg-info=2` only.
Usefull log look like this one 

`21:43:44:788 TY_DATA_REPORT: seq 160, dpid: 0x01, type: 0x04, length: 1, val: 0 `

traschke, 2023-12-09:
I can confirm adding it to `_TZE200_TS0601_trv.json` works for the clone device  `_TZE200_p3dbf6qs` @Smanar mentioned earlier.

More tuya datapoints for this device I found here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/340417a57f5a97550773407e80ca914a17582f78/src/devices/tuya.ts#L3004

I added one for the offset (dpid 47) in my local environment, but cannot yet confirm it works. I think there is still something wrong with the parsing. I copied the code from another TRV (`_TZE200_h4cgnbzg`, different model, not a clone of this one here) I tried the offset to get working, so the parse function probably is mismatching.

In the code of the repo from the link above there is also a [conversion function](https://github.com/Koenkk/zigbee-herdsman-converters/blob/05470a25918c635090a50fe8539b5695b5808f19/src/lib/tuya.ts#L563), but I had not the time to look into it for now.

```json
{
  "name": "config/offset",
  "parse": {
    "dpid": 47,
    // maybe needs fixing
    "eval": "Item.val = Attr.val * 100;",
    "fn": "tuya"
  },
  "default": 0
}
```

Smanar, 2023-12-09:
Ha yep  the "convertion function" is just for negative number, but can be native too.

traschke, 2023-12-09:
Fiddled around a little bit more with the offset, I think I found the correct config.
Using a factor of 10 for offset increased the temperature by 10° (25° with offset 0°, 35° with offset 1°), so I changed the factor to 100 (25° with offset 0°, 26° with offset 1°).

REST API response

No offset:
```json
{
  "config": {
    "heatsetpoint": 2100,
    "locked": false,
    "mode": "heat",
    "offset": 0,
    "on": true,
    "reachable": true
  },
  "etag": "d1de6da4d4766d93b86e7a2fb54cd48f",
  "lastannounced": null,
  "lastseen": "2023-12-09T19:01Z",
  "manufacturername": "_TZE200_p3dbf6qs",
  "modelid": "TS0601",
  "name": "Thermostat Küche",
  "state": {
    "lastupdated": "2023-12-09T19:00:15.566",
    "lowbattery": null,
    "temperature": 2500
  },
  "type": "ZHAThermostat",
  "uniqueid": "a4:c1:38:82:03:39:f9:a6-01-ef00"
}
```

Offset 2°:
```json
{
  "config": {
    "heatsetpoint": 2100,
    "locked": false,
    "mode": "heat",
    "offset": 200,
    "on": true,
    "reachable": true
  },
  "etag": "a6846622dba6f1e371c9a2744d410afb",
  "lastannounced": null,
  "lastseen": "2023-12-09T19:03Z",
  "manufacturername": "_TZE200_p3dbf6qs",
  "modelid": "TS0601",
  "name": "Thermostat Küche",
  "state": {
    "lastupdated": "2023-12-09T19:03:15.653",
    "lowbattery": null,
    "temperature": 2700
  },
  "type": "ZHAThermostat",
  "uniqueid": "a4:c1:38:82:03:39:f9:a6-01-ef00"
}
```

Using this config for offset for now:

```json
{
  "name": "config/offset",
  "parse": {
    "fn": "tuya",
    "dpid": 47,
    "eval": "Item.val = Attr.val * 100;"
  },
  "write":{
    "fn":"tuya",
    "dpid": 47,
    "dt": "0x2b",
    "eval": "Item.val / 100;"
  },
  "read": {
    "fn": "none"
  }
}
```

derdon999, 2023-12-31:
Offset seems not to work for me or at least not as intended. How about the other functionallities like auto or "scenes" for days and times? Is this working for you @traschke ?

Smanar, 2023-12-31:
>Offset seems not to work for me or at least not as intended

What is the result on your side ?
There is an "auto" mode nope ?
For "dailies"  requests, long time we have give up thoses one on tuya stuff. Depend of requests.

>Is there any deconz admin following this or integrating the device in a next release? 

Not me, I can make a PR if you want but I need the compete tested DDF.

traschke, 2023-12-31:
The offset is working for me, I'm using it in my setup in daily use since my last post and I did not notice any issues. I can provide the full ddf I'm using in the next days if @Smanar wants to open a PR.
Did you remove and re-pair the device after updating the ddf? I think deconz does not auto update existing devices after you change the ddf.
For the schedule options with the auto mode, I do not use them. I have more specific rules in my Smarthome setup (OpenHAB in my case) where I control the setpoint temperature based on other events, with the trv set to heat mode when I need the radiator to heat. I also have a rule in OpenHAB to auto adjust the offset value based on another temperature sensor in the room, so the trv temperature somewhat matches the temperature of the external sensor. From that I can confirm the offset works for me.

derdon999, 2024-01-02:
Thanks, maybe the full DDF would help as I might have something wrong in my setup. I will try your DDF. Did you @traschke see more details in Phoscon compared to the screenshot I posted?

derdon999, 2024-01-02:
> What is the result on your side ?

I miss some functionalities specially for deconz/Phoscon. Right now I´m having rules in openhab triggering the state "OFF" or "HEAT" with an external temp sensor (I guess similar to traschke´s config) but I need to set the temperature on the TRV to 35°C (this is max) while Phoscon shows me values ranging from 25°C to 28°C. The max temperature in Phoscon to adjust is 30°C. I have put the TRV to a underfloor heating, but this is not the reason for my issues. I just wanted to explain my constellation and the higher temperature value measured by the TRV. Also the measured values shown on the TRV are not the one shown in Phoscon, which is strange but I guess it´s due to some offset config I tried.

> There is an "auto" mode nope ?

"config/mode": {
            "auto": 0,
            "heat": 1,
            "off": 2
          }

Smanar, 2024-01-02:
>while Phoscon shows me values ranging from 25°C to 28°C. 

You can make rules using OpenHab ? It's a better application.
So your problem is the temperature displayed on third app ? The "offset" we are using is not just an offset (that add or minus a value) but an internal value that impact the regulation too.
Perhaps it work only for positive value and not negative ?

derdon999, 2024-01-14:
Thanks @traschke for providing the DDF. I was playing around a little bit last week and found the same way, as you provided it. Before I had some changes in heatsetpoint as well, which didn´t worked for the offset. 
Now I can confirm, that your offset parameters as provided here work as well for me. I can adjust the offset in openhab, but not in phoscon. Thats ok for now. The only little "problem" I have is, that on the device itself I can adjust the heatsetpoint from 5°C to 35°C and in openhab or phoscon just until 30°C. But now with the working offset that is not an issue anymore, as I don´t need the full range anymore.
Did you see any changes to Phoscon compared to the screenshot in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7333#issuecomment-1804284634 ?

Thanks again @traschke  and @Smanar for the support!

Are there any chances, that Deconz will add the offset function for all users and improve phoscon? As I have the feeling, that the support for new devices is missing, I´m reading that many users are already switching to zigbee2mqtt and I might also.

Smanar, 2024-01-14:
Can you share the error message when trying to set it over 30°C.
From the DDF the limit is 32

```
  "range": [
    500,
    3200
  ]
```

And from code too

`rItemDescriptors.emplace_back(ResourceItemDescriptor(DataTypeInt16, QVariant::Double, RConfigHeatSetpoint, 500, 3200));`

>Are there any chances, that Deconz will add the offset function for all users and improve phoscon?

For deconz and other third app, just need to make a PR with this DDF, if someone want to make it ? (I can if needed)
But we haven't impact on phoscon Web application

derdon999, 2024-01-14:
Can you tell me, where you see that in the DDF?

>   "range": [
>     500,
>     3200
>   ]

Or do I need to add it to heatsetpoint? Can you please explain as I don´t see it.

traschke, 2024-01-14:
I will look into the contributing process and open a PR in the following days for the DDF.

Smanar, 2024-01-15:
>Can you tell me, where you see that in the DDF?

Long story ^^.
Device are first handled by legacy code, so values are limited by `rItemDescriptors.emplace_back(ResourceItemDescriptor(DataTypeInt16, QVariant::Double, RConfigHeatSetpoint, 500, 3200));`
But if there is a DDF, the DDF take the priority. a DDF take his defaut setting in this file for "config/heatsetpoint" https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/generic/items/config_heatsetpoint_item.json
And that you put in the DDF take the priority on value on the "defaut" file.

So you can increase this lmit in the DDF with adding a "range" field.

But the limit need to be 32 not 30, It's for that I don't understand.

traschke, 2024-01-19:
Hey @Smanar, I opened the PR #7537, maybe you can review it? Unfortunately I wasn't able to set the correct tags (Enhancement and Device Improvement), maybe I missed something.

PR #7537 (2024-01-31): DDF for Avatto TRV16 and TRV06 TRVs + add support for offset (_TZE200_TS0601)
Fixes #7333

Adds support for the following TRVs:
* Avatto TRV16 (`_TZE200_rxntag7i`): https://aliexpress.com/item/1005005850177778.html
* Avatto TRV06 (`_TZE200_p3dbf6qs`): https://aliexpress.com/item/1005006191834430.html

In addition, support for the `config/offset` property of the `_TZE200_TS0601` TRVs is added.
closed 2024-01-31 by manup