dresden-elektronik/deconz-rest-plugin #7334: Device Request: SONOFF TRVZB Thermostat

fox34, 2023-11-02:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

TRVZB

### Manufacturer

SONOFF

### Model identifier

TRVZB

### Device type to add

Thermostat

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off
[image]
[image]

### Thermostat
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2023-11-03:
Hello, here you have a defaut DDF for thermostat

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "TRVZB",
  "vendor": "SONOFF",
  "product": "TRVZB Thermostat",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint",
          "default": 0
        },
        {
          "name": "config/mode"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "state/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0201",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 30,
          "max": 360,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}
```

But your device use too the on/off cluster, it have something special ?

fox34, 2023-11-05:
Thank you very much. Now the Thermostat does show up in the Phoscon sensors list and in the REST-API after re-pairing. Controlling the temperature and on/off state works. The readonly OnOff state in the On/Off cluster apparently represents whether a temperature is selected or the thermostat is set to "OFF" (antifreeze).

Smanar, 2023-11-05:
>The readonly OnOff state in the On/Off cluster apparently represents whether a temperature is selected or the thermostat is set to "OFF" (antifreeze).

Do you need it ?
It seem using state/on for thermostat cause issue. The config/mode is working ? it have the "off" state too ?

fox34, 2023-11-14:
> Do you need it ? 

Since OnOff is read-only and mirrored by the Cluster "Thermostat", Attribute 0x001C, System Mode (Auto <-> Off), I don't think it's needed.

> The config/mode is working ? it have the "off" state too ?

config/mode is working, it also has the "off" state. Controlling the thermostat with Homebridge deCONZ generally works as intended (On/Off, reading and setting temperature).

Smanar, 2023-11-14:
So what is missing ?

fox34, 2023-11-14:
Nothing, as far as I can tell. Thank you!

Smanar, 2023-11-14:
If all is working do you want to make the PR or want I make it ? (to add it officialy)

fox34, 2023-11-15:
Created PR: #7357 
Checks seemingly failing because of other DDFs.

Smanar, 2023-11-15:
>Checks seemingly failing because of other DDFs.

Yep, don't worry for that, if there is a real problem, a dev will react. Thx for the PR ^^

closed 2023-12-07 by fox34

Smanar, 2024-01-10:
@fox34 there is somes changes asked in the PR.