dresden-elektronik/deconz-rest-plugin #7335: Device Request: Tuya micro motion sensor ZY-M100-24G - _TZE204_ijxvkhd0 - TS0601

tw090, 2023-11-03:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

ZY-M100 Series Breathing Presence Sensor

### Manufacturer

_TZE204_ijxvkhd0

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

tw090, 2023-11-14:
How can i support to add the device?

tw090, 2023-11-20:
IT IS possible to convert/use this json?
https://github.com/Koenkk/zigbee2mqtt/issues/18237

BabaIsYou, 2023-11-28:
Hi,

Could you try this DDF please : 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE204_ijxvkhd0",
  "modelid": "TS0601",
  "product": "Tuya Human Presence Detector",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/dark"
        },
        {
          "name": "config/tholddark"
[... 201 lines cut ...]
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "tuya"
          },
          "parse": {
            "dpid": 105,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    }
  ]
}
```

tw090, 2023-11-29:
> Hi,
> 
> Could you try this DDF please ? For lux to be updated into GUI you also need lux_to_lightlevel.js to be but in the right directory, or the DDF to be put directly into tuya subdir of your current DecONZ default DDF folder.
> 
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "_TZE204_ijxvkhd0",
>   "modelid": "TS0601",
>   "product": "Tuya Human Presence Detector",
>   "sleeper": false,
>   "status": "Gold",
>   "subdevices": [
>     {
>       "type": "$TYPE_LIGHT_LEVEL_SENSOR",
>       "restapi": "/sensors",
>       "uuid": [
>         "$address.ext",
>         "0x01",
>         "0x0400"
>       ],
>       "items": [
>         {
>           "name": "attr/lastseen"
>         },
>         {
>           "name": "attr/manufacturername"
>         },
>         {
>           "name": "attr/modelid"
>         },
>         {
>           "name": "attr/name"
>         },
>         {
>           "name": "attr/swversion",
>           "read": {
>             "at": "0x0001",
>             "cl": "0x0000",
>             "ep": 1,
>             "fn": "zcl:attr"
>           },
>           "parse": {
>             "at": "0x0001",
>             "cl": "0x0000",
>             "ep": 1,
>             "fn": "zcl:attr",
>             "script": "tuya_swversion.js"
>           }
>         },
>         {
>           "name": "attr/type"
>         },
>         {
>           "name": "attr/uniqueid"
>         },
>         {
>           "name": "config/on"
>         },
>         {
>           "name": "config/reachable"
>         },
>         {
>           "name": "state/lastupdated"
>         },
>         {
>           "name": "state/dark"
>         },
>         {
>           "name": "config/tholddark"
[... 203 lines cut ...]
>           "name": "state/presence",
>           "read": {
>             "fn": "tuya"
>           },
>           "parse": {
>             "dpid": 105,
>             "eval": "Item.val = Attr.val;",
>             "fn": "tuya"
>           },
>           "default": false
>         }
>       ]
>     }
>   ],
>   "bindings": [
>     {
>       "bind": "unicast",
>       "src.ep": 1,
>       "cl": "0xEF00"
>     }
>   ]
> }
> ```

Thank you for the info with lux, now everything works with your DDF!

BabaIsYou, 2023-11-29:
👍 ready to make a pull request to share it in next release ?

tw090, 2023-11-29:
> 👍 ready to make a pull request to share it in next release ?

From my side - yes!

BabaIsYou, 2023-11-29:
> From my side - yes!

Done !

closed 2023-11-29 by manup