dresden-elektronik/deconz-rest-plugin #7336: TUYA TV05-ZG (TRV)

m8ichael, 2023-11-03:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

TV05-ZG

### Manufacturer

VALE (TUYA)

### Model identifier

TS0601

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
/
/

Mimiix, 2023-11-03:
Please add all screns. Also make sure to put the correct manufacturer name + model ID in the opening post.

m8ichael, 2023-11-03:
> Please add all screns. Also make sure to put the correct manufacturer name + model ID in the opening post.

Sorry, but no more Clusters are displayed.

m8ichael, 2023-11-03:
Information on using the attributes can be found [here](https://github.com/Lederhaut/zha_quirks/blob/main/ts0601_trv_zonnsmart.py) as part of another project.

Mimiix, 2023-11-03:
Looks good now. Thanks :)

m8ichael, 2023-11-03:
> Looks good now. Thanks :)
> 
> Could you please add the model ID / manu name in the post?

Done!

Smanar, 2023-11-03:
Hello, can try this DDF to start

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_mudxchsu",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Tuya TRV",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"auto": 0, "heat": 1, "off": 3}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lowbattery",
          "parse": {"fn": "tuya", "dpid": 35, "eval": "Item.val = Attr.val != 0"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 40, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 40, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
              ["auto", 0], ["heat", 1], ["off", 3]
          ],
          "parse": {"fn": "tuya", "dpid": 2, "eval": "if (Attr.val == 0) { Item.val = 'auto' } else if (Attr.val == 1) { Item.val = 'heat' } else { Item.val = 'off' }"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x30", "eval": "if (Item.val == 'auto') { 0 } else if (Item.val == 'heat') { 1 } else { 3 }"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "config/windowopen_set",
          "parse": {"fn": "tuya", "dpid": 8, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 8, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 24, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

m8ichael, 2023-11-04:
Well, I adapted the code somewhat analogously to the referenced parallel project. The temperature and mode are now set correctly - switching off does not yet work, nor do the offset settings for the temperature:

```php
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_mudxchsu",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Tuya TRV",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "meta": {
        "values": {
          "config/mode": {"auto": 0, "heat": 1, "holiday": 2, "holidayready": 3}
        }
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lowbattery",
          "parse": {"fn": "tuya", "dpid": 35, "eval": "Item.val = Attr.val != 0"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/heatsetpoint",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/locked",
          "parse": {"fn": "tuya", "dpid": 40, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 40, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/mode",
          "values": [
              ["auto", 0], ["heat", 1], ["holiday", 2], ["holidayready", 3]
          ],
          "parse": {"fn": "tuya", "dpid": 2, "eval": "if (Attr.val == 0) { Item.val = 'auto' } else if (Attr.val == 1) { Item.val = 'heat' } else if (Attr.val == 2) { Item.val = 'holiday' } else if (Attr.val == 3) {Item.val = 'holidayready'}"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x30", "eval": "if (Item.val == 'auto') { 0 } else if (Item.val == 'heat') { 1 } else if (Item.val='holiday') { 2 } else {3}"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "config/windowopen_set",
          "parse": {"fn": "tuya", "dpid": 8, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 8, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 24, "eval": "Item.val = Attr.val * 10;"},
          "read": {"fn": "none"}
        }
      ]
    }
  ]
}
```

Smanar, 2023-11-04:
>the offset settings for the temperature

Here there is an easy thing to do, all managed by deconz
```

        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 24, "eval": "Item.val = Attr.val * 10 + R.item('config/offset').val;"},
          "read": {"fn": "none"}
        }
```

>switching off does not yet work

Its not using the "mode" on your device ? It's perhaps the dpid 107
```
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 107, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 107, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
```

Else something you can do is looking logs with flag "info" and "info_l2" and turn manualy the device off, you will see the request on log.

m8ichael, 2023-11-04:
Hello,

I think that the offset value not only influences the display, but also determines internally in the device whether the heating needs to be activated or not (heating curve). But I did not yet find the right parameter.

In fact, I cannot deactivate the device directly, but the minimum temperature can be set to 5°C.

However, there is probably a parameter that deactivates the heating mode:

```php
0x0128  # [0] unlocked [1] child-locked --> works
0x0108  # Window-Detection [0] inactive [1] active --> to be tested
0x0210  # [0,0,0,210] target room temp (decidegree) --> works
0x0218  # [0,0,0,200] current room temp (decidegree) --> works
0x0223  # [0,0,0,98] battery charge --> to be tested
0x0402  # [0] Scheduled/auto [1] manual [2] Holiday [3] HolidayReady --> works
0x016B  # Heating Stoppig [0] inactive [1] active --> to be added
0x0265  # BOOST mode operating time in (sec) --> to be added
```

m8ichael, 2023-11-04:
Now I'm irritated: I just wanted to test the changes, now the API shows me a completely different output:

```
"config": {
        "heatsetpoint": 1700,
        "offset": 0,
        "on": true,
        "reachable": true,
        "schedule": {
        },
        "schedule_on": false
    },
    "etag": "e597af8b21aa4838a40a03275262687e",
    "lastannounced": null,
    "lastseen": "2023-11-04T08:34Z",
    "manufacturername": "_TZE200_mudxchsu",
    "modelid": "TS0601",
    "name": "HeizungTest",
    "state": {
        "lastupdated": "2023-11-04T08:30:04.239",
        "on": false,
        "temperature": 2040
    },
    "type": "ZHAThermostat",
    "uniqueid": "50:32:5f:ff:fe:55:93:fd-ff-ef00"
``` 

Is it possible to find out which JSON config file was selected here?

m8ichael, 2023-11-04:
Ah, ok, the "state/on" was the problem. If I remove this entry, then it works again.

Smanar, 2023-11-04:
>I think that the offset value not only influences the display, but also determines internally in the device whether the heating needs to be activated or not (heating curve). But I did not yet find the right parameter

So it's not an offset but an hysteresis ?
I don't see this setting on the device, there is a "local temperature calibration" but this description is for me just an offset, but nevermind you can use it too.

```
        {
          "name": "config/offset",
          "parse": {"fn": "tuya", "dpid": 27, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 27, "dt": "0x2b", "eval": "Item.val;"},
          "read": {"fn": "tuya"}
        },
```

But idk the value convertion

>The minimal value is -5 and the maximum value is 5 with a step size of 0.1

Can you change this value manualy on the device ? to see return on deconz log ?

And for the off IDK why state/on cause issue, can use for test config/interfacemode

```
        {
          "name": "config/interfacemode",
          "parse": {"fn": "tuya", "dpid": 107, "eval": "Item.val = Attr.val;"},
          "write": {"fn": "tuya", "dpid": 107, "dt": "0x10", "eval": "Item.val;"},
          "read": {"fn": "none"}
        },
```

m8ichael, 2023-11-05:
Hi,

> So it's not an offset but an hysteresis ? I don't see this setting on the device, there is a "local temperature calibration" but this description is for me just an offset, but nevermind you can use [...]

Yes, the range +/- 5°C is also mentioned in the device instructions. In what format must the parameter be specified?

> Can you change this value manualy on the device ? to see return on deconz log ?

Unfortunately not, the value is simply transferred and cannot be queried on the device itself. Only the heating control then works “sooner/later”.

> And for the off IDK why state/on cause issue, can use for test config/interfacemode
>
> ```
>         {
>           "name": "config/interfacemode",
>           "parse": {"fn": "tuya", "dpid": 107, "eval": [...]

Are the parameters here 0 and 1? No parameters are accepted here. :-(

Smanar, 2023-11-05:
>Yes, the range +/- 5°C is also mentioned in the device instructions. In what format must the parameter be specified?

From that I m reading it need to be *10, but idk how it will react for negative value.
As deconz work with *100 value, need to send 50 for 0.5 degrée, and divide it again by 10 to have a * 10.

```
       {
          "name": "config/offset",
          "parse": {"fn": "tuya", "dpid": 27, "eval": "Item.val = Attr.val * 10;"},
          "write": {"fn": "tuya", "dpid": 27, "dt": "0x2b", "eval": "Item.val / 10;"},
          "read": {"fn": "tuya"}
        },
```

m8ichael, 2023-11-05:
Hello everyone,

after a day of testing, I can see that the device works extremely unreliably with the settings discussed here. Although the temperature can obviously be set, this temperature change is only partially reflected in the device itself - although deconz always reports back a "success". The same applies to changing the mode. That's not really good. Unfortunately, there still seems to be a need for development here.

Best regards
Michael

closed 2023-12-04 by github-actions[bot]