dresden-elektronik/deconz-rest-plugin #7339: Unsupported device type ZHAParticulateMatter

oywino, 2023-11-07:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

This error message comes every time I restart Home Assistant

### Steps to reproduce the behavior

1. Goto settings --> systemes 
2. Then click on Logs
3. The error message listed below under additional context comes every time.

The term ZHAParticulateMatter has no hits on Google search except 1 hit in a HomeBridge thread on GitHUB. But the maintainer refused to accept the issue and just marked is as "invalid" and closed it.

Is this related to ZHA ? I don't even have the ZHA integration and I have never had it.
Is it related to Matter? I don't have a matter integration and I don't have any Matter devices.

### Expected behavior

Don't know

### Screenshots

Logger: pydeconz.models
Source: components/deconz/gateway.py:356
First occurred: 10:04:47 PM (1 occurrences)
Last logged: 10:04:47 PM

Unsupported device type ZHAParticulateMatter

### Environment

- Host system: QNAP NAS Container Station
- Running method: Docker Container
- Firmware version: 26780700
- deCONZ version 2.24.1
- Device: (Conbee II
- Do you use an USB extension cable: yes 
- Is there any other USB or serial devices connected to the host system? No

### Additional context

Home Assistant Log:

Logger: pydeconz.models
Source: components/deconz/gateway.py:356
First occurred: 10:04:47 PM (1 occurrences)
Last logged: 10:04:47 PM

Unsupported device type ZHAParticulateMatter

SwoopX, 2023-11-07:
I'm afraid this has nothing to do with deconz, therefore closing.

closed 2023-11-07 by SwoopX

oywino, 2023-11-08:
How can it NOT have anything to do with deConz when the error message says:

Logger: pydeconz.models
Source: components/deconz/gateway.py:356

What else do I have to guide me in my search for a solution?