dresden-elektronik/deconz-rest-plugin #7340: Merten / Schneider Dimmer 1 GANG

jensflorian, 2023-11-08:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

MEG5116-0300 Taster with MEG5171-0000 Inset

### Manufacturer

Schneider Electric

### Model identifier

1GANG/DIMMER/1

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off
[image]

### Level Control
[image]

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
[image]

### Schneider Specific
[image]

Smanar, 2023-11-09:
```
{
   "schema":"devcap1.schema.json",
   "manufacturername":"Schneider Electric",
   "modelid":"1GANG/SWITCH/1",
   "product":"1GANG/SWITCH/1",
   "sleeper":false,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_ON_OFF_LIGHT",
         "restapi":"/lights",
         "uuid":[
            "$address.ext",
            "0x01"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"state/alert",
               "description":"The currently active alert effect.",
               "default":"none"
            },
            {
               "name":"state/on",
               "description":"True when device is on; false when off.",
               "refresh.interval":5
            },
            {
               "name":"state/reachable"
            }
         ]
      },
      {
         "type":"$TYPE_SWITCH",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x15",
            "0x0006"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/buttonevent",
               "description":"The last received button event."
            },
            {
               "name":"state/lastupdated"
            }
         ]
      }
   ]
}
```

Hello so same question than on the forum, do you need the button event ? Most wired switch don't have them, because the light state is enought ?

Second question do you need bind ? For light device it's easy, just use it manualy, the state is updated or not ? If not add in the DDF
```

  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x01"
        }
      ]
    },
```

It's a bind + report code for level control and on/off.

jensflorian, 2023-11-09:
Sorry, my first DDF attempt was wrong, it should be:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "1GANG/DIMMER/1",
  "product": "1GANG/DIMMER/1",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
``` 
the State is updated without the bind and Report code. But I think it is useful anyway. Is this then 1 the correct endpoint?

```
,
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
``` 

But I cannot change the DDF - the DDF editor seems to be read-only? What should I do?
Using "Open" also doesn't work

And yes, buttonevents are very useful in home automation such as OpenHab, they can be used as triggers.

Smanar, 2023-11-10:
>Is this then 1 the correct endpoint?

Nope, look your capture, the endpoint is the number on node "titlebar", so you are using the 0x03 for the "light" device

>But I cannot change the DDF - the DDF editor seems to be read-only? What should I do?

You are using the "preview" tab ? so yes this is read only, you can use the editor on previous tab to change values, or better to avoid surprise (the editor can add/remove usefull thing) can use a simple text editor.

>This device here is different (1GANG/DIMMER/1") it is a dimmer!

Have edited my post for dimmer device

>buttonevents are very useful in home automation

But you can use ligh return instead of switch return, this feature is most used for decoupled switch, the physical switch have no control on light and used for other thing.

And you probably will need more bind to use switch return. You can add it on the buttonmap file. But if it don't work just look at log in deconz/help/debug view with flag" info" and "info_l2", if you are seen nothing, you can add

```
    {
      "bind": "unicast",
      "src.ep": 21,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 21,
      "dst.ep": 1,
      "cl": "0x0008"
    },
```

It's a bind from the switch to the coordinator for cluster 0x0008 and 0x0006

jensflorian, 2023-11-10:
Thank you, I managed to update and load the DDF.
I'm still missing buttonevents, so I added your last binds and get the following:
[image]

Is src.ep 21 correct? Because I have no node titlebar with the number 21?  Or is it just hex for node 15?
Do you have other binds I could test?

Smanar, 2023-11-11:
Yep deconz show number in hexa, but the DDF work with decimal number, so 0x15 = 21.

Group are another thing (and your devices support them too), can be this kind of bind
```
    {
      "bind": "groupcast",
      "src.ep": 2,
      "cl": "0x0008",
      "config.group": 0
    }
```

But with the complete DDF you need to see buttonevent in log.
Just enable deconz log with help/debuf view with flag "info" and "info_l2", press the button and get log about missing button map. It will help to create the code in the buttonamp.json file

I think you will probably use the same than for 
```

        "wiserFugaBatteryMap": {
            "vendor": "Schneider Electric",
            "doc": "Wiser LK FUGA battery 4 button switch",
            "modelids": ["FLS/AIRLINK/4", "FLS/SYSTEM-M/4"],
            "buttons": [
                {"S_BUTTON_1": "Top left"},
                {"S_BUTTON_2": "Top right"},
                {"S_BUTTON_3": "Bottom left"},
                {"S_BUTTON_4": "Bottom right"}
            ],
            "map": [
                [2, "0x15", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [2, "0x15", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [2, "0x16", "ONOFF", "ON", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [2, "0x16", "ONOFF", "OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [2, "0x15", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Dim up"],
                [2, "0x15", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [2, "0x15", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Dim down"],
                [2, "0x15", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [2, "0x16", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "Dim up"],
                [2, "0x16", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [2, "0x16", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_4", "S_BUTTON_ACTION_HOLD", "Dim down"],
                [2, "0x16", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"]
            ]
        },
```

But with different endpoint.

Smanar, 2023-11-11:
BTW I have perhaps find a device clone, take a look on this one, he have same enbpoint and same feature (but more gangs, so more code in ddf)

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/wiser/fuga_dimmer_switch.json

```
        "wiserFugaDimmerMap": {
            "vendor": "Schneider Electric",
            "doc": "Wiser LK FUGA dimmer switch",
            "modelids": ["LK Dimmer"],
            "buttons": [
                {"S_BUTTON_1": "Top left"},
                {"S_BUTTON_2": "Top right"},
                {"S_BUTTON_3": "Bottom left"},
                {"S_BUTTON_4": "Bottom right"}
            ],
            "map": [
                [2, "0x15", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [2, "0x16", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [2, "0x15", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Dim up"],
                [2, "0x15", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [2, "0x16", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Dim down"],
                [2, "0x16", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [2, "0x17", "SCENES", "RECALL_SCENE", "1", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Scene 1"],
                [2, "0x18", "SCENES", "RECALL_SCENE", "2", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Scene 2"]
            ]
        },
```

jensflorian, 2023-11-11:
Strange, with groupcast bind and DDF set to gold I don't get any buttonevents. So I removed the DDF completely and just added
the device in buttomap.json:
```
        "wiserFugaDimmerMap": {
            "vendor": "Schneider Electric",
            "doc": "Wiser LK FUGA dimmer switch",
            "modelids": ["LK Dimmer", "1GANG/DIMMER1"],
``` 

Now I'm getting buttonevents for 1002 and 1001. The others give an error. My press sequence was: Short press top, long press top, short press down, long press down.

```
14:09:45:695 [INFO] - Button 1002 - 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 95
14:09:45:741 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x03, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001001, zclSeq: 96
14:09:47:838 [INFO] - Button 1001 - 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: LEVEL_CONTROL (0x0008), action: Dim up, payload: 0046, zclSeq: 97
14:09:48:212 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x03, cluster: LEVEL_CONTROL (0x0008), command: 0x0A, payload: 000020D5, zclSeq: 98
14:09:48:381 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: LEVEL_CONTROL (0x0008), command: STOP (0x03), payload: 0000, zclSeq: 99
14:09:49:211 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x03, cluster: LEVEL_CONTROL (0x0008), command: 0x0A, payload: 000020F9, zclSeq: 100
14:09:50:404 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 101
14:09:50:431 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x03, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001000, zclSeq: 102
14:09:52:492 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: LEVEL_CONTROL (0x0008), command: MOVE (0x01), payload: 01460000, zclSeq: 103
14:09:53:738 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: LEVEL_CONTROL (0x0008), command: STOP (0x03), payload: 0000, zclSeq: 104
14:09:54:586 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 105
``` 

Two questions: Are buttonevents generated by deconz itself, when a device is joined to a group?
And why kills a functioning DDF the buttonevents (info_L2 is then completely empty)?

Smanar, 2023-11-11:
buttonevent are generated by the device.

```
14:09:52:492 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: LEVEL_CONTROL (0x0008), command: MOVE (0x01), payload: 01460000, zclSeq: 103
14:09:53:738 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: LEVEL_CONTROL (0x0008), command: STOP (0x03), payload: 0000, zclSeq: 104
```

Can try with adding
```

                [2, "0x15", "LEVEL_CONTROL", "MOVE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "aaaa"],
                [2, "0x15", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "bbbb"],
```

`14:09:50:404 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: ONOFF (0x0006), command: OFF (0x00), payload: None, zclSeq: 101`

For this one

`[2, "0x15", "ONOFF", "OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "cccc"],`

Hard to see the action you have done
But as you have only 1 gang need to use only S_BUTTON_1 and for the action, just need to see the "action" in log will be "aaaa" or "bbb" ....

jensflorian, 2023-11-12:
Trying to combine switch and dimmer into a single buttonmap (same zigbee button module):

```
        "wiserDimmerMap": {
            "vendor": "Schneider Electric",
            "doc": "Wiser Merten dimmer switch",
            "modelids": ["1GANG/DIMMER/1", "1GANG/SWITCH/1"],

            "map": [
                [2, "0x15", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [2, "0x15", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [2, "0x15", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Dim up"],
                [2, "0x15", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [2, "0x15", "LEVEL_CONTROL", "MOVE", "0", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Dim down"],
                [2, "0x15", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"]
            ]
        },
```

Only dimming down is missing. There are two Messages when pressing and releasing:
```
11:31:45:824 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x03, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001001, zclSeq: 69
11:31:48:643 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x03, cluster: LEVEL_CONTROL (0x0008), command: 0x0A, payload: 00002004, zclSeq: 71
```
What is missing from above buttonmap. json ?

Smanar, 2023-11-12:
>It doesn't work with "S_BUTTON_1" or "S_BUTTON_2"

How many physical button have your device ? Buttonevent are X00Y form, the X is the button number and Y the event.

For the dimming try adding (For the release, more usefull)

`[2, "0x03", "LEVEL_CONTROL", "0x0A", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "xxxxx"],`

jensflorian, 2023-11-14:
> For the dimming try adding (For the release, more usefull)
> 
> `[2, "0x03", "LEVEL_CONTROL", "0x0A", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "xxxxx"],`

This one triggers exactly every  5min. I think this is some special communication intended for the Wiser hub (which I do not have).

Smanar, 2023-11-14:
And this one ?

> 11:31:45:824 [INFO] - No button handler for: 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x03, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001001, zclSeq: 69

jensflorian, 2023-11-14:
> And this one ?
I think all from 0x03 are permanent triggers. But I think, I now found the missing one for dimming down:
```
18:21:42:805 [INFO] - Button 2001 - 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: LEVEL_CONTROL (0x0008), action: Dim down, payload: 01460000, zclSeq: 114
18:21:43:966 [INFO] - Button 2003 - 1GANG/DIMMER/1, unicast to: 0x0000, endpoint: 0x15, cluster: LEVEL_CONTROL (0x0008), action: Stop, payload: 0000, zclSeq: 116
``` 

Here is my working buttonmap:

```
        "wiserDimmerMap": {
            "vendor": "Schneider Electric",
            "doc": "Wiser Merten dimmer switch",
            "modelids": ["1GANG/DIMMER/1", "1GANG/SWITCH/1"],

            "map": [
                [2, "0x15", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [2, "0x15", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [2, "0x15", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Dim up"],
                [2, "0x15", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [2, "0x15", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Dim down"],
                [2, "0x15", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"] 
            ]
        },
``` 

For the DDF I removed the groupcast to keep traffic in the network low. I think you can commit this now.

Smanar, 2023-11-14:
>For the DDF I removed the groupcast to keep traffic in the network low. I think you can commit this now.

Sure, you don't want to make the PR yourself ? Else no problem, I can do it (if you can share the last DDF).

jensflorian, 2023-11-14:
> Else no problem, I can do it (but if you can share the last DDF, with your last modifications).

That would be great, I have no experience with committing. Here is the DDF.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "1GANG/DIMMER/1",
  "product": "1GANG/DIMMER/1",
  "sleeper": false,
  "status": "Silver",
  "path": "/devices/dimmer1.json",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x03"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ZGP_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x15",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/buttonevent",
          "description": "The last received button event."
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
"bindings": [
    {
      "bind": "unicast",
      "src.ep": 21,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 21,
      "dst.ep": 1,
      "cl": "0x0008"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x01"
        }
      ]
    }
  ]
}
```

Smanar, 2023-11-15:
Here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7356
Have made some changes, for exemple have removed "1GANG/SWITCH/1" as this one have DDF yet.

closed 2023-11-28 by manup