dresden-elektronik/deconz-rest-plugin #7342: Merten / Schneider Dimmer 2 GANG

jensflorian, 2023-11-08:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

MEG5126-0300 Taster with MEG5172-0000 dual Insel

### Manufacturer

Schneider Electric

### Model identifier

2GANG/DIMMER/2

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]

### Level Control
[image]

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
[image]

jensflorian, 2023-11-08:
[image]

Smanar, 2023-11-14:
Now the 1 gangs is finished, I think the 2 gangs will be faster ^^.
Just need to copy/paste/rename the 0x15 part in 0x16 for the second gangs

jensflorian, 2023-11-14:
Unfortunately I'm having problems with buttonevents. While the switche is shown in phoscon, I get following messages in debug log:

```
19:57:18:139 don't create binding for attribute reporting of sensor 2GANG/DIMMER/2 127
19:57:18:140 don't create binding for attribute reporting of sensor 2GANG/DIMMER/2 128
``` 

Anything wrong with the DDF?

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "2GANG/DIMMER/2",
  "product": "2GANG/DIMMER/2",
  "sleeper": false,
  "status": "Silver",
  "path": "/devices/dimmer2.json",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x03"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
[... 219 lines cut ...]
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 4,
      "cl": "0x0008",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}```

Smanar, 2023-11-15:
>don't create binding for attribute reporting of sensor

You can have this message only for device using Legacy code, mean your device don't use DDF core, try with status="Gold"
Else your bind table seem fine for me.

You haven't error about buttonmap when you use it on logs ?

jensflorian, 2023-11-15:
> You haven't error about buttonmap when you use it on logs ?

Yes, no error messages, even with gold Status. The light states are reported correctly.

Smanar, 2023-11-16:
You can try without the goupcast part, but you need to have something, you have trie with "gold" status ?

jensflorian, 2023-11-16:
Ah, the double ZHA switch was most likely the problem. I also removed the groupcast part. I'm not getting buttonevents and I figured out the remaining maps for 0x16.

```
19:56:21:863 [INFO] - Button 4001 - 2GANG/DIMMER/2, unicast to: 0x0000, endpoint: 0x16, cluster: LEVEL_CONTROL (0x0008), action: Dim down, payload: 01FF0000, zclSeq: 125
```

The DDF
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "2GANG/DIMMER/2",
  "product": "2GANG/DIMMER/2",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/dimmer2.json",
  "subdevices": [
    {
      "type": "$TYPE_DIMMABLE_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x03"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert",
          "description": "The currently active alert effect.",
          "default": "none"
        },
        {
          "name": "state/bri",
          "description": "The current brightness.",
          "refresh.interval": 5
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
[... 172 lines cut ...]
            "vendor": "Schneider Electric",
            "doc": "Wiser Merten dimmer switch",
            "modelids": ["1GANG/DIMMER/1", "1GANG/SWITCH/1", "2GANG/DIMMER/2"],

            "map": [
                [2, "0x15", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [2, "0x15", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [2, "0x16", "ONOFF", "ON", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [2, "0x16", "ONOFF", "OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [2, "0x15", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Dim up"],
                [2, "0x15", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [2, "0x15", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Dim down"],
                [2, "0x15", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [2, "0x16", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "Dim up"],
                [2, "0x16", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [2, "0x16", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_4", "S_BUTTON_ACTION_HOLD", "Dim down"],
                [2, "0x16", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"]

            ]
        },
```  

I'm pretty sure this will also work with the double switch (which I dont have, but expect to be 2GANG/SWITCH/2 from the current notation).

I think you can commit this as well. Thank you very much for your help!

Smanar, 2023-11-17:
>I'm not getting buttonevents

But it works now or not ?

jensflorian, 2023-11-17:
Sorry, my bad, that should have been a „now“. Yes, it works!

Smanar, 2023-11-18:
Added on same PR too https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7356

closed 2023-11-28 by manup