dresden-elektronik/deconz-rest-plugin #7346: Conbee III no firmware

yves19, 2023-11-11:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I have installed a new Conbee III stick on my computer USB2 in place of a Conbee II one.
With Phoscon I have loaded my previously saved configuration as mentioned in the Phoscon documentation.
But the ConbeeIII key is not reachable (firmware not mentioned in deconz nor in Phoscon)

### Steps to reproduce the behavior

1 : plug the Conbee III stick
2 : start deconz application
3 : try to connect the ConBeeIII
4 : after a while the connection is in error

The error is systematic.

### Expected behavior

Conbee III shall connected after few seconds in deconz

### Environment

- Host system: ( PC)
- Running method: ( Windows10)
- Firmware version: (unknown)
- deCONZ version : (2.23.1)
- Device: (ConBee III )
- Do you use an USB extension cable: (yes ) 
- Is there any other USB or serial devices connected to the host system? Ino

### deCONZ Logs

14:41:40:046 wait reconnect 13 seconds
14:41:41:045 wait reconnect 12 seconds
14:41:42:050 wait reconnect 11 seconds
14:41:43:037 wait reconnect 10 seconds
14:41:43:693 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
14:41:44:045 wait reconnect 9 seconds
14:41:44:728 networkState: CC_ReconnectNetwork
14:41:45:043 wait reconnect 8 seconds
14:41:46:043 wait reconnect 7 seconds
14:41:47:046 wait reconnect 6 seconds
14:41:48:036 wait reconnect 5 seconds
14:41:49:043 wait reconnect 4 seconds
14:41:49:063 COM: \\.\COM5 : USB Serial Port (0x0403/0x6015)
14:41:49:064 GW firmware update select USB device
14:41:49:065 GW update firmware not found: 
14:41:50:048 wait reconnect 3 seconds
14:41:51:048 wait reconnect 2 seconds
14:41:52:041 wait reconnect 1 seconds
14:41:52:045 COM: \\.\COM5 : USB Serial Port (0x0403/0x6015)
14:41:52:111 auto connect com \\.\COM5
14:41:53:697 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
14:41:59:059 GW update firmware not found: 
14:42:02:567 device disconnected reason: 1, device index: 0
14:42:03:041 wait reconnect 15 seconds
14:42:03:044 COM: \\.\COM5 : USB Serial Port (0x0403/0x6015)
14:42:03:696 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
14:42:04:049 wait reconnect 14 seconds
14:42:05:038 wait reconnect 13 seconds
14:42:06:036 wait reconnect 12 seconds
14:42:07:037 wait reconnect 11 seconds
14:42:08:045 wait reconnect 10 seconds
14:42:09:035 wait reconnect 9 seconds
14:42:09:055 COM: \\.\COM5 : USB Serial Port (0x0403/0x6015)
14:42:09:056 GW firmware update select USB device
14:42:09:057 GW update firmware not found: 
14:42:10:042 wait reconnect 8 seconds
14:42:11:047 wait reconnect 7 seconds
14:42:12:037 wait reconnect 6 seconds
14:42:13:045 wait reconnect 5 seconds
14:42:13:701 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
14:42:14:036 wait reconnect 4 seconds
14:42:15:038 wait reconnect 3 seconds
14:42:16:043 wait reconnect 2 seconds
14:42:17:047 wait reconnect 1 seconds

bachph, 2023-11-11:
I might have a similar problem on Ubuntu 20.04

`lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal`

Looks like it's recognized correctly by the system

`[181068.242706] usb 1-2: new full-speed USB device number 3 using xhci_hcd
[181068.399763] usb 1-2: New USB device found, idVendor=0403, idProduct=6015, bcdDevice=10.00
[181068.399768] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[181068.399770] usb 1-2: Product: ConBee III
[181068.399771] usb 1-2: Manufacturer: dresden elektronik
[181068.399773] usb 1-2: SerialNumber: DE0311XXXX
[181068.403300] ftdi_sio 1-2:1.0: FTDI USB Serial Device converter detected
[181068.403337] usb 1-2: Detected FT-X
[181068.404264] usb 1-2: FTDI USB Serial Device converter now attached to ttyUSB0`

But in the Phoscon App its not recognized:

`Vendor: dresden elektronik
Product: RaspBee
Gateway Version: 2.24.1
App Version:
Firmware: Not connected`

Any idea? Tried the stable and the beta version.

yves19, 2023-11-11:
Eureka !

With beta version 2.24.1 of deconz the Conbee II works. Older versions of deconz do not support Conbee III.

bachph, 2023-11-11:
> Eureka !
> 
> With beta version 2.24.1 of deconz the Conbee II works. Older versions of deconz do not support Conbee III.

So you got the Conbee III running on Windows with beta 2.24.1?

I have 2.24.1 running on Ubuntu, but the Conbee III is not recognized correctly.

yves19, 2023-11-11:
Yes for test purpose i begin on win 10 on a PC  and Conbee III works as expected.

I will for the next step perform tests on a NUC running Debian 11.

Pedda83, 2023-11-12:
> > Eureka !
> > With beta version 2.24.1 of deconz the Conbee II works. Older versions of deconz do not support Conbee III.
> 
> So you got the Conbee III running on Windows with beta 2.24.1?
> 
> I have 2.24.1 running on Ubuntu, but the Conbee III is not recognized correctly.

I have the same issue with 2.24.0 and 2.24.1. Conbee III could not connected. I run it on docker.

Smanar, 2023-11-12:
You don't have another machine to try it ?

Pedda83, 2023-11-12:
> You don't have another machine to try it ?

i start it with an completly new docker container without any date of my prod-system. no luck. i've install it on windows 11 and it runs. than i install an virtualbox on windows 11 with the same ubuntu 20.04 like my docker and install deconz locally - it runs.

Spy-1, 2023-11-13:
same her on docker:
`COM: /dev/ttyUSB1 :  (0x0000/0x0000) COM: /dev/ttyUSB0 :  (0x0403/0x6015) failed open com status: (-3), path: /dev/ttyUSB0`

Mimiix, 2023-11-13:
No clue what this is. But this means device is not selected / accessible.

Locking this up, as we are going offtopic.

closed 2023-11-13 by Mimiix