dresden-elektronik/deconz-rest-plugin #7347: TRADFRI bulb E27 WS opal 980lm non-latin characters

rikroe, 2023-11-12:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

In https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7213 / https://github.com/dresden-elektronik/deconz-lib/commit/1940240e8560a656c3e2b7d72f15d137d3520d39 some changes were made to accomodate latin-1 characters in *Manufacturer Name*.

I have updated to 2.24.1, but the DDF for this light is not found. It seems as if there is still some issue with the latin-1 to ASCII conversion:
```
09:50:52:005 DEV no DDF for 0x000B57FFFE4FD06A, modelId: TRADFRI bulb E27 WS�opal 980lm
```

### Steps to reproduce the behavior

1. Update to 2.24.1
2. Check the logs if you have an TRADFRI bulb E27 WS opal 980lm (firmware version 0x23050631)
3. No DDF is found for device

### Expected behavior

DDF is found and no � character in the device name.

### Environment

- Host system: PC
- Running method: Docker container
- Firmware version: 0x26400500
- deCONZ version (not Home assistant Addon version!): 2.24.1
- Device: ConBee I
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so: Which?

ebaauw, 2023-11-12:
Did you try and read the _Basic_ cluster attributes in the GUI?  The light reports the weird character in the _Read Attributes Response_, but deCONZ should filter that out, showing only regular characters in the _Model Identifier_.  That should match the DDF file.

rikroe, 2023-11-12:
After some more tries I have solved the issue by updating the bulb firmware from 2.3.050 (0x23050631) to 2.3.095 (0x23095631).
Now, after a Docker container restart, the DDFs for the updated bulbs are recognized correctly for the updated bulbs:
```
11:48:39:531 DEV found DDF for 0x000B57FFFE4FD06A, path: /usr/share/deCONZ/devices/ikea/tradfri_bulb_e27_ws_opal_980lm.json
...
11:48:39:542 dlg action: Read binding table
...
11:48:39:572 DEV found DDF for 0x000B57FFFE8D3A2E, path: /usr/share/deCONZ/devices/ikea/tradfri_bulb_e27_ws_opal_980lm.json
...
11:48:39:575 DEV no DDF for 0x000B57FFFE4C01F4, modelId: TRADFRI bulb E27 WS�opal 980lm
11:48:39:575 DEV create on-the-fly DDF for 0x000B57FFFE4C01F4
```

Only issue was to figure out what the correct firmware version is, as there is no 980lm version directly and I had to figure out the hex > decimal conversion for the image type...
For reference, this bulb needs image type 0x2002 or 8706 and I have installed the following firmware: [159695-2.1-TRADFRI-bulb-ws-1000lm-2.3.095.ota.ota.signed](http://fw.ota.homesmart.ikea.net/global/GW1.0/01.20.065/bin/159695-2.1-TRADFRI-bulb-ws-1000lm-2.3.095.ota.ota.signed).

If I read the **Basic** cluster, it shows the correct name. However when starting up, I get the message above.

[image]

I would have one additional bulb with the old firmware lying around, but not sure if it is worth the effort to figure out more if the fix is just "do the OTA update".

rikroe, 2023-11-13:
Closing this as it was solved by updating the bulb firmware. Although it introduced some other (maybe unrelated issues in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7302#issuecomment-1807614459).

closed 2023-11-13 by rikroe