dresden-elektronik/deconz-rest-plugin #7349: Tradfri LED bulb E14 470 lumen

MSL-DA, 2023-11-13:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tradfri LED bulb E14 470 lumen

### Manufacturer

IKEA of Sweden

### Model identifier

TRADFRIbulbE14WSglobeopal470lm

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off

[image]

### Level Control

[image]

### Color Control

[image]

ebaauw, 2023-11-13:
What’s the _Image Type_ of the light’s firmware?

MSL-DA, 2023-11-13:
> What’s the _Image Type_ of the light’s firmware?

Do you mean the firmware version for the light?

ebaauw, 2023-11-13:
No, the _Image Type_, the identifier for the firmware.  See the OTAU plugin (select the light and press _Query_), or read the attributes of the _OTAU_ cluster (and select another cluster, and re-select _OTAU_ to work around a GUI bug not showing the values after read).

MSL-DA, 2023-11-13:
> What’s the _Image Type_ of the light’s firmware?

OTAU

[image]

ebaauw, 2023-11-13:
Attribute 0x0008, but it doesn't report it here.  Better check the OTAU panel.

MSL-DA, 2023-11-13:
> Attribute 0x0008, but it doesn't report it here. Better check the OTAU panel.

You need to be more specific in your requests. I do not understand?

ebaauw, 2023-11-13:
See https://phoscon.de/en/support#ota-update-osram-devices

MSL-DA, 2023-11-13:
Nothing happens. Note that Homebridge/Homebridge deCONZ is on LAN 192.168.20.20 and deCONZ is published via docker on 192.168.10.4

MSL-DA, 2023-11-15:
Is this what you are looking for?

[image]

ebaauw, 2023-11-15:
Yes, thanks

MSL-DA, 2023-11-16:
I can add model number LED2002G5 and product code 504.867.87

MSL-DA, 2023-12-07:
^^

MSL-DA, 2023-12-29:
^^

closed 2024-01-28 by github-actions[bot]