dresden-elektronik/deconz-rest-plugin #7350: Tradfri LED bulb GU10 345 lumen

MSL-DA, 2023-11-13:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tradfri LED bulb GU10 345 lumen

### Manufacturer

IKEA of Sweden 

### Model identifier

TRADFRIbulbGU10WS345lm

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off

[image]

### Level Control

[image]

### Color Control

[image]

ebaauw, 2023-11-13:
What’s the _Image Type_ of the light’s firmware?

MSL-DA, 2023-11-15:
[image]

MSL-DA, 2023-12-30:
^^

closed 2024-01-29 by github-actions[bot]

djsmix, 2024-05-29:
is this still a problem?