dresden-elektronik/deconz-rest-plugin #7351: IKEA Gunnarp LED Ceiling

MSL-DA, 2023-11-13:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Gunnarp LED Ceiling

### Manufacturer

IKEA of Sweden

### Model identifier

GUNNARP panel round

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off

[image]

### Level Control

[image]

### Color Control

[image]

ebaauw, 2023-11-13:
What’s the _Image Type_ of the light’s firmware?

MSL-DA, 2023-11-13:
OTAU

[image]

MSL-DA, 2023-11-15:
[image]

MSL-DA, 2023-12-07:
^^

MSL-DA, 2023-12-30:
^^

closed 2024-01-29 by github-actions[bot]