dresden-elektronik/deconz-rest-plugin #7352: Hue White Ambiance Candle E14

MSL-DA, 2023-11-13:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Hue White Ambiance Candle E14

### Manufacturer

Philips

### Model identifier

LTW012

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off

[image]

### Level Control

[image]

### Color Control

[image]

ebaauw, 2023-11-13:
What’s the _Image Type_ of the light’s firmware?

MSL-DA, 2023-11-13:
OTAU

[image]

MSL-DA, 2023-11-15:
[image]

bluemoehre, 2023-11-16:
Got such a device here. Wanted to return it due to coil whine. I can generate a DDF via deCONZ, so you don't have to type all the config. Will this help, @ebaauw  ?

ebaauw, 2023-11-16:
I need to check this weekend, but we probably already have a DDF for this firmware type.  In that case, we just need to add the _Model Identifier_ to that DDF.

ebaauw, 2023-11-16:
Hm. this is an old ZLL light.  Existing white ambiant DDF is for ZB3 lights.   I think I might have one of those lying around somewhere.  See if I can find that this weekend.

MSL-DA, 2023-12-11:
If time is short, I will look into it myself. Just say so.

ebaauw, 2023-12-14:
The DDF is relatively simple, copying light_zll_gamut_C and removing out the colour stuff.  I run into one issue though: the light supports the `candle` effect (at least with the latest firmware 1.108.5), but, obviously, not `colorloop`.  Somehow that causes the API not to show `cap/color/effects`, and accept `colorloop` as value for `state/effect`, resulting in API error 901: Internal error, 951 (since the light doesn't support _Color Loop Set_).  Also the API doesn't understand to cancel the `candle` effect when setting `state/effect` to `none`.

I'm increasingly regretting my decision to expose special effects under `state/effect`, combined with `colorloop`.

closed 2023-12-29 by MSL-DA