dresden-elektronik/deconz-rest-plugin #7353: Hue White Ambiance E27

MSL-DA, 2023-11-13:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Hue White Ambiance E27

### Manufacturer

Philips

### Model identifier

LTW001

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off

[image]

### Level Control

[image]

### Color Control

[image]

### OTAU

[image]

MSL-DA, 2023-11-15:
[image]

ebaauw, 2023-11-16:
These are old ZLL lights, that somehow share the firmware image with the gamut-B extended color lights.  I'm guessing/hoping they behave the same as the 0x010C.  I need to check if I have a color temperature light with this firmware build.

ebaauw, 2023-12-08:
Bump.

I'll try and have a look this weekend.

MSL-DA, 2023-12-11:
If time is short, I will look into it myself. Just say so.

ebaauw, 2023-12-14:
I don't think I have any LTW001.  I found the E14 LTW012 from #7352 and an 800lm E27 LTW010, both with image type 0x010C.

Could you please list the Hue Effects cluster?

MSL-DA, 2023-12-15:
Do you still need the Hue Effects cluster?

closed 2023-12-29 by MSL-DA