dresden-elektronik/deconz-rest-plugin #7358: https://deconz.dresden-elektronik.de/ubuntu/beta/ lacks deconz-dev-2.24.*.deb files (latest releases)

bjornfor, 2023-11-17:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

The latest release is 2.24.x, but https://deconz.dresden-elektronik.de/ubuntu/beta/ lacks the *-dev package for it. (The non-dev package is there though.)

### Steps to reproduce the behavior

1. Go to https://deconz.dresden-elektronik.de/ubuntu/beta/
2. See that e.g. deconz-2.24.2-qt5.deb is there but not deconz-dev-2.24.2.deb (the *-dev package).

### Expected behavior

The *-dev package is available at the same time as the non-dev package, as they are versioned together.

### Environment

(Not relevant.)

SwoopX, 2023-11-17:
No bug

closed 2023-11-17 by SwoopX

bjornfor, 2023-11-17:
Are you saying this should go to the forum instead? I thought the forum was for user self-help, but this issue definitely requires developer attention.

SwoopX, 2023-11-17:
What requires developer attention in your view please? You ticked that you want to report that deconz is programatically misbehaving, which is doesn't. And the dev package doesn't exist anymore but this is the wrong place to discuss this 🤷‍♂️

bjornfor, 2023-11-18:
> And the dev package doesn't exist anymore

That is the info I was looking for, and I think only a dev would know, hence I felt the bugtracker was appropriate.

bjornfor, 2023-11-18:
Consider this: The README in *this repo* says to download the latest release from the URL I'm mentioning in this issue. And until recently there was a matching *-dev package, which is now gone. I considered that an issue worth bringing to the attention of the developers.

Now that I read your comment again (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7358#issuecomment-1817234971), I'm not even sure whether you're saying "yes, I can see it's missing but it should be there" or "yes, it is missing since we intentionally stopped making it".

And where is the right place to discuss this if not here?