dresden-elektronik/deconz-rest-plugin #7364: Tuya ZigBee Button Smart Scene Switch

hakano14, 2023-11-20:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Smart Button

### Manufacturer

_TZ3000_qgwcxxws

### Model identifier

TS0041

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

hakano14, 2023-11-22:
This looks very similor as #7205 - maybe this could be solved in the same mannar?!

hakano14, 2023-12-18:
I think this can be fixed by adding
_TZ3000_qgwcxxws to 
_TZ3000_TS0041_1gang_bat.json and button_maps.json.

Someone who can make this changes?

retrography, 2024-01-06:
@hakano14 Can you test this on your side?

hakano14, 2024-01-06:
It works in Api and Phoscon on my side!
But no predefined events in HA (but event works )
1002 - click
1004 - double click

Mimiix, 2024-01-06:
> It works in Api and Phoscon on my side! But no predefined events in HA (but event works ) 1002 - click 1004 - double click

That's because HA needs to add support. Events are fine

hakano14, 2024-02-07:
Hi!

I'm sorry to say there was a typo error in the new version (2.25.2)...

The manufacturername should be **_TZ3000_qgwcxxws** not TZ3000_qgwcxxws as in devices/tuya/_TZ3000_TS0041_1gang_bat.json

In button_maps.json it is correct.

Hope you can fix this for the next release.

Mimiix, 2024-02-07:
Re opened on request.

retrography, 2024-03-01:
This has been resolved and merged.

Mimiix, 2024-03-01:
Great thanks !

closed 2024-03-01 by Mimiix

7R0J4Ner, 2024-03-03:
Hello

Could you add this version of that Switch too?
_TZ3000_axpdxqgu 
[image]

retrography, 2024-03-04:
Submitted as a part of #7633