dresden-elektronik/deconz-rest-plugin #7367: LIDL SILVERCREST Smart Plug

Splash005, 2023-11-21:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Livarno / Silvercrest / Lidl Smart Plug

### Manufacturer

_TZ3000_kdi2o9m6

### Model identifier

TS011F

### Device type to add

Switch

### Node info

Actually I was sure that this Switched used to be supported. It is a smart switch from the german Lidl.

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off
[image]

LeoeLeoeL, 2023-11-27:
In my case, it's supported but Deconz reports a double smartplug.
[image]

BabaIsYou, 2023-11-29:
"Basic" smart switch are usually supported by legacy code. Is there something special with this one that need a specific DDF to be used?

Splash005, 2023-11-29:
Maybe if that is helping. Last year when I remember that I had it working I was running it on a raspberry pi and now it runs on a synology nas docker.

BabaIsYou, 2023-12-01:
May be this kind of DDF should do the trick : 
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_kdi2o9m6",
  "modelid": "TS011F",
  "product": "Another Smart socket plug",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 5,
          "max": 300
        }
      ]
    }
  ]
}
```

closed 2023-12-01 by SwoopX